OctoPrint/OctoPrint #1480: Can't update Octoprint and plugins

battlezone, 2016-09-06:
I can't update Octoprint and its plugins. The log says the update of Octoprint itself was succesful but the Telegram notifications plugin failed. When reloading the web interface, Octoprint 1.2.9 is still installed and it has not updated.

#### What were you doing?

I wanted to update Octoprint as well as all the plugins that had updates by pressing check for updates now and pressing update now on the popup.

#### What did you expect to happen?

All plugins and Octoprint updated to latest version

#### What happened instead?

A message box stated update has failed and I needed to consult the log

#### Branch & Commit or Version of OctoPrint

Octoprint 1.2.9
Telegram Notifications 1.1
TouchUI 0.3.2 

#### Browser and Version of Browser, Operating System running Browser

Opera running on Windows 10

#### Link to octoprint.log

http://pastebin.com/PGqaEaE7

foosel, 2016-09-09:
From your log, it looks like everything but the Telegram plugin just updated fine. Have you tried restarting OctoPrint manually? If an error during one of the updates is detected, OctoPrint does not automatically restart since that might mask the error for the user.

For the update issue with the telegram plugin you should report that in the plugin repository instead.

PS: Always copy-paste the *full* bug template. Do not remove anything from it that is not enclosed in [ and ]. Otherwise the bot will get angry/stay angry.

battlezone, 2016-09-09:
Thank you for your reply. I thought I did restart Octoprint but apparently I didn't, sorry. After restarting Octoprint manually the browser prompted me to reload and it now loaded the new version.

foosel, 2016-09-09:
Perfect. Can this be closed then?

battlezone, 2016-09-09:
Yes, Thanks again.

closed 2016-09-09 by battlezone