OctoPrint/OctoPrint #1481: Not reporting layer print time

Astroman, 2016-09-06:
In Version 1.2.15, layer print time on the gcode viewer page is always reported as "a few seconds." Remaining print time, etc seems ok though. If you can't see this problem yourself, let me know and I'll be glad to provide whatever other info you need. Thanks.

Astroman, 2016-09-06:
#### What were you doing?

Printing an object and looking at the Gcode viewer

#### What did you expect to happen?
See actual layer print time
#### What happened instead?
Layer print time always showed "a few seconds" even though it was much longer
#### Branch & Commit or Version of OctoPrint
1.2.15 (master branch)
[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version
Printrbot simple, not sure about version
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Safari 9.1.3, Mac OS 10.11.6
[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser
Not printing right now, will provide later if necessary
[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:
Not printing right now, will provide later if necessary
[If applicable. Always include if unsure or reporting UI issues.]

foosel, 2016-09-09:
> Layer print time always showed "a few seconds" even though it was much longer

Layer print times - like everything else - are estimates which depending on multiple factors can be more or less accurate. For this reason I switched all print time estimates in 1.2.14/15 to be displayed in a very fuzzy way since people do depend too much on them to be accurate otherwise.

On the `maintenance` branch/future 1.2.16 release I've switched at least the individual layer times in the GCODE viewer back to show as full durations instead of fuzzed up estimates:

[image]

That's all I can do here for now. 

If the layer times in general are off (it says "a couple of seconds" but takes several minutes instead), that's most certainly caused by something in your GCODE files making the estimate in the GCODE viewer go haywire. I'd need an example file to be able to look further into this.

PS: Always copy-paste the *full* bug template. Do not remove anything from it that is not enclosed in [ and ]. Otherwise the bot will get angry/stay angry.

foosel, 2016-09-12:
@Astroman 

> If the layer times in general are off (it says "a couple of seconds" but takes several minutes instead), that's most certainly caused by something in your GCODE files making the estimate in the GCODE viewer go haywire. I'd need an example file to be able to look further into this.

Is this the case or were you only referring to the layer times being stated in a fuzzy manner? Important to know, because if the former there might be more action needed, if the latter this can be considered closed.

Astroman, 2016-09-12:
I looked more carefully at the issue and it appears to say “a few seconds” whenever the layer print time is under a minute. In other words, it never estimates short layer times in seconds. I thought it always said “a few seconds" since I happened to be looking at the gcode viewer only when layer times were under a minute.

That said, I think it used to provide estimates for layer times that were under a minute. Of course, I can’t confirm this, and if this is not correct, I apologize. I seem to remember using these relatively short layer print times of 20-50 seconds or so to estimate the amount of time left to get to a higher layer.

I wonder if you are rounding off or truncating to the nearest minute which would be fine for long layer times but misleading for short ones, espeially those under a minute.

Thanks.

—Dave

foosel, 2016-09-24:
Yes, as I said earlier, 1.2.15 had a fuzzy display in place for the layer times. That's been changed in 1.2.16 though which I released yesterday. In a nutshell, the non-fuzzy layer times are back. Closing this as solved.

closed 2016-09-24 by foosel