OctoPrint/OctoPrint #1482: disconnects after sending one gcode command

raymatos, 2016-09-07:
#### What were you doing?
taz 3 printer
raspberry pi 2 

Steps to reproduce
1. start up raspberry pi
2.Click on connect 
3. After connected, go to control
4. click on one of the arrows - printer will respond
5. Click on another direction arrow
6. Printer disconnects and throws error

#### Branch & Commit or Version of OctoPrint

Version 1.2.15 (master branch)

#### Printer model & used firmware incl. version

Taz 3 marlin 

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 52.0.2743.116 on Mac OS 10.11.4

#### Link to octoprint.log

```
2016-09-07 02:35:46,179 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.10.74
2016-09-07 02:35:58,028 - octoprint.timelapse - INFO - Deleted old unrendered timelapse idler_jig_with_logo_20160831012932
2016-09-07 02:35:59,234 - octoprint.timelapse - INFO - Deleted old unrendered timelapse FilamentClip_20160831003755
2016-09-07 02:35:59,338 - octoprint.timelapse - INFO - Deleted old unrendered timelapse FilamentClip_20160831002824
2016-09-07 02:36:13,089 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.42s
2016-09-07 02:36:25,642 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 8.1.1
2016-09-07 02:36:27,228 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-09-07 02:36:35,784 - octoprint.timelapse - INFO - Deleted old unrendered timelapse ModSpoolTower15_single_20160831023619
2016-09-07 02:38:06,737 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Taz_Openbuild_X_Carriage_1_2_a.gcode
2016-09-07 02:51:27,483 - octoprint.timelapse - INFO - Deleted old unrendered timelapse ModSpoolTower15_single_20160831024512
2016-09-07 02:52:19,630 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.10.74
2016-09-07 02:52:22,609 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.10.74
2016-09-07 02:54:13,831 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.10.74
2016-09-07 02:54:15,349 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.10.74
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                2016-09-07 02:17:45,037 - octoprint.server - INFO - Starting OctoPrint 1.2.15 (master branch)
2016-09-07 02:17:45,225 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-09-07 02:17:50,255 - octoprint.plugin.core - INFO - Found 9 plugin(s) providing 9 mixin implementations, 6 hook handlers
2016-09-07 02:17:50,301 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-09-07 02:17:51,423 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-09-07 02:17:51,910 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-09-07 02:18:03,047 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 8.1.1
2016-09-07 02:18:03,061 - octoprint.plugin.core - INFO - Initialized 9 plugin implementation(s)
2016-09-07 02:18:03,097 - octoprint.plugin.core - INFO - 9 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/pluginmanager
|  Print History Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/softwareupdate
|  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
|  TouchUI (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.15-py2.7.egg/octoprint/plugins/virtual_printer
2016-09-07 02:18:03,109 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-09-07 02:18:03,131 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-09-07 02:18:03,155 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-09-07 02:18:03,161 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-09-07 02:18:06,790 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2016-09-07 02:18:07,188 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-09-07 02:18:07,262 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-09-07 02:18:07,275 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-09-07 02:18:07,305 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-09-07 02:18:08,076 - octoprint.plugins.stlviewer - INFO - STL Viewer loaded!
2016-09-07 02:18:08,587 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.10.100/'}
[... 141 lines cut ...]
2016-09-07 04:08:39,063 - SERIAL - DEBUG - Send: M105
2016-09-07 04:08:39,130 - SERIAL - DEBUG - Recv: ok T:23.6 /0.0 B:25.4 /0.0 @:0 B@:0
2016-09-07 04:08:39,139 - SERIAL - DEBUG - Send: N0 M110 N0*125
2016-09-07 04:08:39,153 - SERIAL - DEBUG - Recv: ok
[... cut ...]
2016-09-07 04:08:54,822 - SERIAL - DEBUG - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1618
2016-09-07 04:08:54,904 - SERIAL - DEBUG - Changing monitoring state from 'Operational' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1618'
2016-09-07 04:08:54,962 - SERIAL - DEBUG - Connection closed, closing down monitor
 
```

foosel, 2016-09-07:
Something is up there with your printer connection, causing it to run out of synch on a level below OctoPrint. There's nothing OctoPrint can do about that sadly.

In the past users ran into this when attempting to access the serial port of the printer from another program at the same time as OctoPrint, but I guess this is not the case here. You could check if another USB cable makes a difference here, maybe even a different (stronger) power supply on the pi (in case it is caused by some weird brown out issue), but sadly that is about all I can recommend right now.

In any case, this is not a bug in OctoPrint. OctoPrint relies on the serial communication between it and the printer working on the transport level, which for whatever reason appears to not be the case here.

It is interesting that the serial goes down right on a probably long move, and if it does do that on a z home every time that might also hint at the cause potentially.

foosel, 2016-09-07:
PS: you didn't copy the full template, hence the bot complained. Marking this as complete manually.

closed 2016-11-08 by foosel