OctoPrint/OctoPrint #1484: load file (https error)

MoonshineSG, 2016-09-08:
#### What were you doing?

load a gcode file 

#### What did you expect to happen?

file to be loaded and shown in the GCODE viewer

#### What happened instead?

JS error 

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0.dev1365+g64d484b (devel branch)

[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version

not relevant

#### Browser and Version of Browser, Operating System running Browser

Chrome OSX

#### Link to octoprint.log

n/a

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

jquery-2.1.4.js:8630 OPTIONS https://i3.*******.***:80/downloads/files/local/9502805a-d438-478b-8a05-c2127e69a738.gcode net::ERR_SSL_PROTOCOL_ERROR

There are 6 other warnings (unrelated, but worth having a look at):

/#:1 Mixed Content: The page at 'https://i3..*******.***/#' was loaded over HTTPS, but requested an insecure image 'http://plugins.octoprint.org/assets/img/plugins/detailedprogress/example-eta.jpg'. This content should also be served over HTTPS.
/#:1 Mixed Content: The page at 'https://i3..*******.***/#' was loaded over HTTPS, but requested an insecure image 'http://plugins.octoprint.org/assets/img/plugins/gcodebar/gcodebar.png'. This content should also be served over HTTPS.
/#:1 Mixed Content: The page at 'https://i3..*******.***/#' was loaded over HTTPS, but requested an insecure image 'http://www.thing-printer.com/wp-content/uploads/2016/04/octoprint-redeem.png'. This content should also be served over HTTPS.
/#:1 Mixed Content: The page at 'https://i3..*******.***/#' was loaded over HTTPS, but requested an insecure image 'http://plugins.octoprint.org/assets/img/plugins/pushover/pushover.png'. This content should also be served over HTTPS.
/#:1 Mixed Content: The page at 'https://i3..*******.***/#' was loaded over HTTPS, but requested an insecure image 'http://plugins.octoprint.org/assets/img/plugins/telegram/features3.png'. This content should also be served over HTTPS.

#### Screenshot(s) showing the problem:

n/a

I have read the FAQ.

foosel, 2016-09-08:
OctoPrint doesn't do SSL termination itself, so there's a reverse proxy in play here that takes care of that.

> https://i3.*******.***:80/

That hints at a configuration mistake in your reverse proxy. I just tested with a correctly configured nginx and that set everything correctly to have the download links be generated correctly as `https://host/...`.

What proxy are you using for termination and could you provide the configuration you used? That would allow to try to reproduce the issue to figure out if it's indeed caused by a misconfiguration or a weird bug after all.

MoonshineSG, 2016-09-08:
Hmmm... I understand, but the funny thing is that everything was fine until I upgraded to "1.3.0.dev1365+g64d484b"
I went back to  "1.3.0.dev1352+g7b6b806" and the problem disappeared. 

Anyway, I am using the octopi default haproxy with some changes:

```
frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/XXX.XXXXX.pem
        option forwardfor except 127.0.0.1
        capture request header Host       len 32
         # Optional, just for statistics:
         capture request header User-Agent len 200
        
        acl prusa hdr(host) -i prusa.XXXXX.XXX
        acl webcam path_beg /webcam/
        acl timelapse path_beg /downloads/timelapse/

        use_backend octoprint if timelapse
        use_backend mjpg_streamer if webcam
        use_backend webcam_ui if prusa
        default_backend octoprint

backend octoprint
        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        reqadd X-Scheme:\ https if { ssl_fc }
        server octoprint 127.0.0.1:5000

backend mjpg_streamer
        reqrep ^([^\ :]*)\ /webcam/(.*) \1\ /\2
        reqadd X-Scheme:\ https if { ssl_fc }
        server mjpg_streamer 127.0.0.1:8080

backend webcam_ui
        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        server mobile_ui  127.0.0.1:88
```

foosel, 2016-09-08:
I guess it's something that haproxy then does differently after all regarding the request headers. In order to fix a bug with login cookies (see #1479) I had to rewrite the reverse proxy integration. I guess something went wrong there with regards to the stock haproxy config and SSL. Will have to investigate.

foosel, 2016-09-12:
Should be fixed in 1.2.16rc1, the `maintenance` and the `devel` branches

MoonshineSG, 2016-09-13:
tested in "devel" and looks good. thanks!

closed 2016-09-13 by MoonshineSG