OctoPrint/OctoPrint #1485: Not showing all temps on flashforge style printer

davidknag, 2016-09-08:
#### What were you doing?

Printing with my left print head, the right print head unheated. (though it happens always) Looking at the temperature graph.

#### What did you expect to happen?
Show the temperatures for all 3 thermistors (right head, left head, heat bed).

#### What happened instead?
Only the right extruder head, listed as 't0', was shown. I'd like to note that on simplify3d T0 is the number for the heated bed, but I'm not sure how/if they convert over when compiled. The bed was heated but no temp was shown.

#### Branch & Commit or Version of OctoPrint
OctoPrint: 1.2.15 

#### Printer model & used firmware incl. version
Monoprice Architect (Flashforge Creator), upgraded with dual extruders and heated bed. I think it runs an atmega2560.

#### Browser and Version of Browser, Operating System running Browser
Chrome Version 52.0.2743.116 m (64-bit)
#### Link to octoprint.log
unsure how to get a log. Please let me know if you'd like to see.

#### Link to contents of terminal tab or serial.log
I mean.. it's printing. so probably only movement gcode in here
http://pastebin.com/gvcVGsQJ

#### Link to contents of Javascript console in the browser
n/a, console looks fine.

#### Screenshot(s) showing the problem:
https://i.imgur.com/nDKNsvm.png

foosel, 2016-09-08:
Please copy the template *completely* when reporting the issue or the bot will get angry.

Regarding the issue, have you configured multiple extruders and the heated bed in your printer profile (Settings > Printer Profiles)? If yes I'd have to request to take this over to the issue tracker of the gpx plugin and @markwal, I'm not familiar how temperatures from multiple extruders are handled by that plugin/on Makerbot clones like the Flashforge and if there might be an issue somewhere there.

In general OctoPrint (like the gcode based firmwares) enumerates extruders starting at T0, with the bed being B.

davidknag, 2016-09-09:
yes I have, I was thinking that may be the case of being an issue with the gpx plugin. Thanks

closed 2016-09-12 by foosel