OctoPrint/OctoPrint #1488: When a print is active, the folder is locked ( can't upload any files )

Eriobis, 2016-09-11:
#### What were you doing?

I am uploading files via samba to the raspberry pi. Everything works fine while not printing

#### What did you expect to happen?

The file to be uploaded as usual

#### What happened instead?

The file is being deleted ( there's no error while dragging the file onto the remote folder )

#### Branch & Commit or Version of OctoPrint

1.2.16rc1

#### Printer model & used firmware incl. version

Prusa I3 MK2, FW version 3.0.8

#### Browser and Version of Browser, Operating System running Browser
Chrome xxxx on Windows 7 ( this isn't browser related )
#### Link to octoprint.log

pastebin.com/g3rkzcJb

#### Link to contents of terminal tab or serial.log

The file is empty

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

foosel, 2016-09-12:
I was going to say that I can't reproduce it when I understood your setup: You didn't share the watched folder via samba, you used an existing samba share as the watched folder, concurrently, from multiple OctoPrint instances. Do I understand this correctly?

Then it's no surprise what you are experiencing. The watched folder works such that when OctoPrint detects that a file was added there, it *moves* it to its upload folder, as in, the file will no longer be in the watched folder after wards. If more than one instances are attempting that concurrently, only the first one will succeed. Why am I deleting the file instead of just copying it over? Because otherwise a heavy used watched folder might turn into a storage sink.

So, if I understand your setup, OctoPrint is working as expected here and you'd need to do this setup differently I fear, eg have dedicated folders on your central share, one per instance. At least currently there's no way to have shared storage across multiple OctoPrint instances without them interfering with each other. Once I get around to add a plugin interface for additional file storages, that might change, but right now things are working fine as they are.

Btw, when reporting tickets for release candidates, please also state if stuff was still working on the stable version (1.2.15 here) - I was a bit puzzled when I got this ticket because I didn't touch the watched folder implementation.

Eriobis, 2016-09-12:
You are right, 
I did changed my settings. I use to have the upload of the 3 instances pointing to the /home/pi/gcodeUploads ( SAMBA shared ), but since I saw your video ( octoprint on air 4 ) and you said it was a bad thing, I switched to the watch folder method instead.
Perhapd I will add subfolders for the 3 printers or some kind of script to copy it in all folders.

Can you specify why the method of having the "uploads" folder of multiple instances pointing to the same folder isn't good ?

Thanks

foosel, 2016-09-12:
>Can you specify why the method of having the "uploads" folder of multiple instances pointing to the > same folder isn't good ?

Because the "uploads" folder does not only hold the files itself but also metadata for the files in .metadata.yaml, and concurrent access to that file will also cause things to break.

And as already said in the above mentioned video, OctoPrint also not just copies uploads in there but also sanitizes the filename to make sure it doesn't cause problems in certain settings - when you move stuff in there manually, that sanitization step is skipped and things might stop working.

In a nutshell: OctoPrint's uploads folder should not be manipulated manually (it will break less easily with 1.2.16, but it's still something OctoPrint expects not to be manipulated externally) since OctoPrint does some record keeping and sanitization in there and expects to have full control over everything that goes on in that folder, including the uploaded files and their metadata - messing with either while OctoPrint is running will cause problems, even if only from other OctoPrint instances. OctoPrint's watched folder can be shared across a network, but should NOT be shared among multiple OctoPrint instances, or they will interfere with each other when processing anything that's copied into it. It should be viewed as a temporary "write-only" folder, which you drop files in to have OctoPrint do its thing with and then delete the copy in the watched folder.

I'm btw surprised that sharing the uploads folder across three instances did not cause spectacular failures, considering that every uploaded file would have hit the GCODE analysis on all three instances and the resulting metadata would have been attempted to write back by three instances near consecutively O_O

Anyhow.. if it's ok with you I'd rather mark this as a feature request (something along the lines of "support a centralized uploads folder shared across multiple OctoPrint instances"). As it is now, it's not a bug but rather a design limitation.

Eriobis, 2016-09-12:
Sure, thanks a lot for your prompt answer.
And no, there hasn't been any spectacular crash. I didn't print the same file on 2 printers at the same time tho. I bet this would have been really bad.
Once I get more familiar with that git hub issue thing, I might be able to pick the right subject for the issue :+1:

bolsoncerrado, 2017-02-08:
I second the request!

When you own a printer farm, centralizing the files is IDEAL! :D this way all prints are away to a click of a button and updated to the very last version at the same time!

NiklasOlofsson, 2017-05-04:
Also having multiple Octoprints running. I would like to use one of the installs to share all files to the other ones. Please make it happen. ;)

musyne, 2019-02-08:
> Well after 2 months I've been sharing one unique UPLOADS folder for all my Pis/Printers and 0 issues so far. Of course, I've never tried to print the same file from 2 different Pis at the same time (if I've ever needed to do so, i just clone the file under a different name and reupload).
> 
> Perhaps not the safest solution, but in the meantime I can share that works quite fine!

How did you do it?
NFS mount from the slave Pi's to master Pi??

bolsoncerrado, 2019-02-09:
I have a shared folder in a network drive and all my pis access it on boot up from fcstab mapping and that's all.

total1ty, 2023-07-07:
Just an idea for a workaround on this issue:

Couldn't I just keep separate upload folders and thus independent metadata for all the octoprint instances, and then linking a centralized folder containing the gcode files via softlinks into each?

That way each instance could keep track of their own metadata, but the gcode files stay synchronized.
I know that strictly speaking, I shouldn't modify the uploaded files externally, but I've always uploaded my gcodes through SMB and never had problems, The Metadata gets generated as soon as I refresh the File Browser it seems.

I'm gonna give this a try and report back.

bolsoncerrado, 2023-07-07:
Be sure to work on a tutorial if it works!  :P  THanks!

kantlivelong, 2023-07-07:
> I second the request!
> 
> When you own a printer farm, centralizing the files is IDEAL! :D this way all prints are away to a click of a button and updated to the very last version at the same time!

A centralized way to manage is ideal and there are solutions for that (OctoFarm/FDM-Monster). Centralized file storage itself is a single point of failure and will bite you _eventually_, OctoPrint itself is not designed to work in a clustered fashion and thus I don't see support for this request ever getting implemented in core.

Given that, if you really want to go this route you could make a plugin that replays file events to other instances (plus fixing inconsistencies). But if you're going to go through the effort of notifying other instances why not just use something already built for the task? Yes those solutions don't use a central file store but storage is cheap.

jneilliii, 2023-07-07:
there are also plugins like onedrive files, googledrive files, and gitfiles that allow you to centralize your files in a cloud service and sync down to the instances configured for it.

bolsoncerrado, 2023-07-07:
Just realized this thread is SIX years old lol

Nothing broke spectacularly (actually NOTHING broke) and we have printed the same file from more than one instance at the same time too.

jneilliii, 2023-07-07:
> ONLY issue we have had so far is the slicer thumbnails not appearing on all instances...

just realized that issue got closed, will re-open.

kantlivelong, 2023-07-07:
> So what is googledrive, Dropbox or github but just another NAS type with the added requisite of being online?

The plugins revolving around these services _sync_ the data to the OctoPrint instance. Syncing will trigger file change events just like the `watched` folder. For Linux systems inotify does not work on cifs/nfs mount points so we would have to resort to periodic scans of the `uploads` dir and perform data comparison/hasing which can be problematic and slow over cifs/nfs with large files especially when being accessed in a farm. Currently the `watched` folder is periodically scanned but anything placed there is simply ingested.

> But aside from that, no issues at all with file corruption nor traffic jam not anything I could think of.

Right, it works fine and will likely work fine until one day something happens. It's really just poor design to have a single point of failure when a machine can be compartmentalized. Sure having to go to every machine individually is a nightmare but that's where tools to manage that come in. 

> (actually NOTHING broke)
Does this ticket actually need to be open then?

bolsoncerrado, 2023-07-08:
> Right, it works fine and will likely work fine until one day something happens. It's really just poor design to have a single point of failure when a machine can be compartmentalized. Sure having to go to every machine individually is a nightmare but that's where tools to manage that come in.

I won't say "it can't break" bc you're the code gurus here but 6 years without issues sound reliable to me so far :P  

> > (actually NOTHING broke)
>  Does this ticket actually need to be open then?

I still hope one day to have a NAS solution  for octoprint haha! :P whatever it is, plugin, autoupdate, cronwhatever or something "mortals" can do just PnP, fire'n'forget and such ;)