OctoPrint/OctoPrint #1495: No system menu in the navigation bar

malnvenshorn, 2016-09-13:
Hi,

after adding the commands for shutdown, reboot and restarting OctoPrint I wondered where I can activate those commands. After some search I found out that there should be a pulldown menu in the navigation bar, but this menu is missing. All I can see are settings, the user menu and the announcements icon. Of course the specific user has admin rights. I'm using version 1.2.15 of OctoPrint and opened the web frontend with Firefox 48.0.2 and InternetExplorer 11. Same result with both browsers.

foosel, 2016-09-15:
If you haven't, try restarting OctoPrint after adusting config.yaml. If you have already done that and it didn't help, please get in touch on either the mailing list or the g+ community. When you do, also include your config.yaml file or at least the snippet you copy-pasted into it. It's sadly impossible to help you without having access to that.

closed 2016-09-15 by foosel

malnvenshorn, 2016-09-15:
Yes I already restarted OctoPrint and I haven't touched the config.yaml file. I set up everything inside the web interface under Settings > Server > Commands. The generated config.yaml file can be found here: https://paste.debian.net/823918/

foosel, 2016-09-16:
Ah, that explains it. While current `devel` already uses the commands from Settings > Server > Commands to also to populate the system menu, current stable releases still need those things [configured separately](http://docs.octoprint.org/en/master/configuration/config_yaml.html#system) and the commands are only used for internal functions such as the software update process there.

Try:

``` yaml
system:
  actions:
  - name: Restart OctoPrint
    action: restart
    command: sudo systemctl restart octoprint.service
    confirm: You are about to restart OctoPrint.
  - name: Reboot the server
    action: reboot
    command: sudo reboot
    confirm: You are about to reboot the whole server.
  - name: Shutdown the server
    action: shutdown
    command: sudo poweroff
    confirm: You are about to shutdown the whole server.
```

in your `config.yaml` and things should work on 1.2.x as well. You can also leave out the `confirm` lines if you don't want a confirmation dialog.

ctgreybeard, 2016-09-16:
I'm running 1.2.16rc2 and I tried adding those commands under Settings->Server->Commands but they do not appear in the System menu, even after restarting the service.

foosel, 2016-09-16:
`devel` is 1.3.0dev. 1.2.16rc2 fits the above mentioned version pattern 1.2.x and still needs the `system.actions` stanzas in config.yaml as outlined above.

ctgreybeard, 2016-09-16:
Ah! Thanks.

jcdammeyer, 2016-09-24:
I've attached my Replicape config.yaml file.   I now do have the System menu up there but none of the actions have any effect.  It doesn't shut down or reboot.  Also I've noticed that there's no direct link with the fields in the server menu to the System menu selections so there are really two issues.  

if I perform a sudo poweroff on the command line a short while later my Beagle is off.  But from the menu nothing.  Same with the -r restart or restarting octoprint.

I've looked at the web source for yaml syntax and there's no linkage so is that done in the Python application then  that links the commands in the fields to the menu entries?
Thanks,
[Uploading Replicape-config.yaml.txt…]()