OctoPrint/OctoPrint #1496: Feature Request: Display OctoPrint's IP on the LCD after connection.

BigBox3D, 2016-09-14:
We have a farm of 20+ BigBoxes and various other printers dotted around the network. It can be quite difficult to find the right printer to connect to.

It would be very helpful to have OctoPrint send an M117 command to display the current IP on the LCD after connection.

Salandora, 2016-09-14:
This should either be able to do by a plugin or maybe even by the EventSystem and a script wich then uses the cli of octoprint.
Hmm I gonna check that and hopefully give you a solution in the next days.

KC703, 2016-09-17:
Hope you're using static IP addresses for that farm... if you are then labels and asset tags are probably the way to go.  :-)  Has to be easier than reading an IP address on the LCD screen.

BigBox3D, 2016-09-17:
It's not just for the farm, but also for new machines as they are turned on for the first time, or if the DHCP is reset.

BillyBlaze, 2016-09-17:
Why not setup a [zero-config](https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#reach-your-printer-by-typing-its-name-in-address-bar-of-your-browser---avahizeroconfbonjour-based) and connect with printer1.local, printer2.local etc.

Then take a number sticker and put them on the printer so you can see on the other side of the room what printer you need to connect with.

jneilliii, 2016-09-24:
I'll be posting to the Google Groups for a little help, but I have the basic framework in place for a plugin that will do this (in theory).  I have it now where it can retrieve the ip and write it to the log file on printer connection, but I can't get the plugin hook [octoprint-comm-protocol-scripts](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-scripts) to add the M117 command to the afterPrinterConnected gcode script.  You can find the plugin over [here](https://github.com/jneilliii/OctoPrint-ipOnConnect).

jneilliii, 2016-09-24:
So with a little direction from [markwal](https://github.com/markwal) I think I got this plugin working now.  I don't have an LCD panel attached but it is working with my other M117 plugins ([navbar](https://github.com/jneilliii/OctoPrint-M117NavBar) and [popup](https://github.com/jneilliii/OctoPrint-M117PopUp)). You can find the plugin [here](https://github.com/jneilliii/OctoPrint-ipOnConnect).

BigBox3D, 2016-09-27:
Excellent, thanks :)

dattas, 2016-10-20:
This is also in my DetailedProgress plugin (version 0.1.4) and works great on my BigBox Pro 1.1. @BigBox3D If one of these plugins work for you, it would be great to close this issue (to help Gina keep the issue tracker clean!)

foosel, 2016-11-30:
I'm going to go out on a limb here and consider this close since there is not only one but two plugins that take care of that.

closed 2016-11-30 by foosel