OctoPrint/OctoPrint #1497: Feature Request: Docking Script Button.

BigBox3D, 2016-09-14:
In the control tab have a button similar in looks to the home buttons but one which launches a custom script, editable in the settings GCODE Scripts page.

The purpose is to execute GCODE which can be used to park the head in a particular location and or perform a nozzle wiping cycle before parking.

Salandora, 2016-09-14:
Uhm you know that you can define parametriezed gcodes and add buttons to the control tab yourself? 
You'll find how on the really nice documentation page somwhere here: http://docs.octoprint.org

BigBox3D, 2016-09-15:
I did not. I shall look into it. Thanks :)

foosel, 2017-01-27:
Definitely possible via custom controls, closing for now. If you are missing something from what's possible via custom controls for your specific use case, feel free to request a reopen (and explain ;)).

closed 2017-01-27 by foosel