OctoPrint/OctoPrint #1498: TypeError when printing x3g on flashforge creator pro

wrigjl, 2016-09-15:
#### What were you doing?

Printing an x3g file to my printer.

#### What did you expect to happen?

Printed object.

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

1.2.15 master branch

#### Printer model & used firmware incl. version

FlashForge Creator Pro (Sailfish 7.7)

#### Browser and Version of Browser, Operating System running Browser

Firefox/Linux, Chrome/MacOS

#### Link to octoprint.log
#### Link to contents of terminal tab or serial.log

https://gist.github.com/wrigjl/af297b2dd4678d1b72fb70a9c0e46b01

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

 State
Machine State: Offline: TypeError: 'must be string without null bytes, not str' @ comm.py:_do_send_without_checksum:2031
File: Spacer.x3g 
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: - 
Printed: - / 25.1KB

I have read the FAQ.

zip file attached is the object being printed.  Printing it from SD works fine.

[Spacer.zip](https://github.com/foosel/OctoPrint/files/475151/Spacer.zip)

foosel, 2016-09-15:
@markwal can you take a look?

markwal, 2016-09-15:
Yep. GPX plugin doesn't know how to print x3g directly. You can only print x3g from the printer's sd card.  Octoprint and GPX allow you to upload and print gcode.

One day, I'll figure out how to both allow the x3g to be listed to printed, but give this info in an error message when trying to upload x3g to OctoPrint or gcode to the printer's SD card. But I'll track that on the plugin's issue tracker.

closed 2016-09-15 by markwal

foosel, 2016-09-15:
Thanks, I remembered something along the lines of that but want sure if I wasn't mixing stuff up.

wrigjl, 2016-09-15:
Oops, sorry for the bother... Sure enough using the untranslated gcode is fine.

JulianGoesPro, 2017-01-20:
hey, sorry for waking this up, but does that mean I can't start a x3g file print via octoprint at all? (want to time laps it via octoprint)

Thanks!

markwal, 2017-01-21:
You can start an x3g print from OctoPrint. However, you have to copy it to the printers SD card yourself (sneaker net) and timelapse needs to be a time interval, not z-change.