OctoPrint/OctoPrint #1499: Vertical Alignment of Progress Bar Text Not Centered

jneilliii, 2016-09-17:
#### What were you doing?

Printing a part.

#### What did you expect to happen?

Progress bar text to display properly with middle vertical alignment.

#### What happened instead?

1.2.16rc2 

#### Branch & Commit or Version of OctoPrint

Version: 1.2.16rc2 (rc/maintenance branch)

#### Printer model & used firmware incl. version

Ordbot Hadron running Marlin 

#### Browser and Version of Browser, Operating System running Browser

Version 53.0.2785.116 m (64-bit) on Windows 10 64 Bit.

#### Link to octoprint.log
[octoprint.log](https://gist.githubusercontent.com/jneilliii/a28d6e32acbe280ba92899e95142978d/raw/6d42520f888e4ce71706bd23b19101d633326e5c/octoprint.log)

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

[javascript console](https://gist.githubusercontent.com/jneilliii/5e5488fe4f87c5d283541f051634c7fc/raw/40af9a655934f33d5d78aeee365b01493f8a6b3c/javascript.console)

#### Screenshot(s) showing the problem:

[progresstext](https://cloud.githubusercontent.com/assets/5249455/18606156/000388b8-7c75-11e6-84e0-02a121ad3d95.PNG)

I have read the FAQ.

markwal, 2016-09-24:
Huh.  That's the same browser (Google Chrome) and version I'm using and it doesn't show that way for me.  Weird.

jneilliii, 2016-09-24:
Yeah I don't know.  Maybe one of my installed plugins or chrome extension is messing something up.  I updated to Version: 1.2.16 (rc/maintenance branch) stable version this morning and it appears the same for me.

jneilliii, 2016-09-25:
I also can't remember if the text was dark when the background area was white, and then become white once the blue reached it, but it's not doing that for me either.  I've tried disabling all my chrome extensions, in the middle of the print so obviously can't remove my installed plugins.

[image]

foosel, 2016-09-26:
Cannot reproduce that here either. Chrome version is 53.0.2785.116 m as well.

Could you try with a fresh Chrome profile, to rule out any Chrome extensions being at fault here?

jneilliii, 2016-09-26:
I tried disabling all the extensions and clearing my cache and it was the same for me.  Let me check from another user profile and see if it's still messed up.

jneilliii, 2016-09-27:
So I've looked from 3 different computers with the same results.  Opening up developer tools and highlighting the text element on the progress bar it's showing with a computed vertical-align css property of baseline.  Here's a screenshot of what I mean.

[image]

jneilliii, 2016-09-27:
I just tested in a fresh install of Firefox and am seeing the same thing.

foosel, 2016-10-05:
The `baseline` value is ok. The whole line should be as high as the progress bar, and not have any top or bottom padding:

[image]

[image]

----

To compare, this is the styles applied to the various parts of the progress bar over here:

### `<div class="progress progress-text-centered">...</div>` (the wrapper around everything)
[image]

### `<span class="progress-text-back">...</span>` (the dark text on the unfilled bar)
[image]

### `<div class="bar">...</div>` (wrapper of the filled part)
[image]

### `<span class="progress-text-front">...</span>` (the white text on the filled bar)
[image]

Any obvious differences? Still cannot reproduce, also tested Firefox and even IE and Safari, looks as it should everywhere

foosel, 2017-02-24:
Closed due to lack of response from the OP :/

closed 2017-02-24 by foosel

jneilliii, 2017-02-26:
Sorry for not responding.  It's not really that big of a deal since it's only visual.  What I see on my end is that the containing div seems to be double-height for some reason, and therefore the text isn't aligning the same.  

**Here's what I see on the page:**
[image]

**With the code:**
[image]

**and CSS for that div:**
[image]

markwal, 2017-02-26:
One of your plugins is overriding the styles.  It's not this style though that is causing the problem, but perhaps one of the nested ones.  You can see in your CSS for that div that "position: relative" is coming from packed_plugins.css, however it should have no effect since it is overriding "position: relative" which is the same.  But perhaps another involved style.  What plugins do you have installed again?

jneilliii, 2017-02-26:
These are what I have enabled right now.  I'll try disabling them all again and restarting and add them individually.

Custom Control Editor (0.2.1)
Flash Arduino (0.1) by Pim Rutgers  
M300Player (0.1.0)
TouchUI (0.3.6)

jneilliii, 2017-02-26:
I have just confirmed this is related to a plugin.  Once I uninstalled all of them and restarted everything does appear as it should.  I'll try to determine which plugin is causing the issue and report a bug with the developer.

foosel, 2017-02-26:
As a quick heads up, I recently noticed that the CSS of core and plugins was being merged, causing the core styles to also be in the plugins file and hence these kind of overrides to appear. I've since fixed that on maintenance and devel and will roll this out with 1.3.2. But as already mentioned, that override should not be causing a problem since it's exactly the same style.

Might be some weird side effect though with plugin CSS, maybe together with the merge issue.

jneilliii, 2017-02-26:
The issue did appear to be related to the flashArduino plugin.  I submitted a pull request to it's css file to hopefully resolve.  In the meantime, I cloned the repository and made the proposed change myself and it does seem to work now without breaking the default progress bars.