OctoPrint/OctoPrint #1501: Solidoodle 2 print in the wrong direction => hit upper right at start/ but controls are OK

patrick95470, 2016-09-18:
HI
Does anybody has the some problem, I wanna use Octopi with a Solidoodle SD2, I can connect it without problem, move the chariot with the control , but when I start the printing it hurst the upper right.

I somebody could help me it would be great :-)

Kind regards

Patrick

patrick95470, 2016-09-18:
HI Again

Let me complete my initial request

I'm using the version Version: 1.2.15 (master branch) of Octoprint

The printer is a Solidoodle 2 1st generation, 

The Browser is safari, chrome or Firefox, whatever the browser , the problem remains the same

I was ready to fill out a bug report , but the URL is broken
404: Page not found – the page https://goo.gl/GzkGv9 does not exist.
If you typed in or copied/pasted this URL, make sure you included all the characters, with no extra punctuation.

So to sum up, I'd like to know what I have to change in my settings for having this nice software driving properly my 3D printer ;-)

I can move the chariot with the control , but when I start the printing is hurting the upper right

Patrick

Salandora, 2016-09-18:
Hello @patrick95470,

Could you please try same gcode files from a PC with any other host software?
I believe it has nothing todo with Octoprint.
Because Octoprint is just streaming what is inside your file.
It would also help if you show us one of your code files. Maybe you somehow reset the XY coordinates inside your start code or so. Dunno.

patrick95470, 2016-09-18:
Hi,

thx for your reply, it works with repetier or mattercontrol with the gcode file sent

 so maybe it is the cura plugin .... when I upload the stl file to slice it into gcode with cura ?

Patrick

patrick95470, 2016-09-18:
Hi,

Just let me update the situation, when I sent a GCODE file, it is OK

The problem is when I use a STL file that I want to slice...

As my 3D Printer is a Solidoodle 2, does somebody can tell me where I can found slice profils to import ?

Kind regards

Patrick

ntoff, 2016-09-20:
You need Cura on your PC. Create a slicing profile on your PC all nicely tuned which you then export from Cura on the PC and import into Octoprint. Just make sure you adhere to the Cura version limits outlined here: https://github.com/foosel/OctoPrint/wiki/Plugin:-Cura

patrick95470, 2016-09-20:
Thx,

Patrick

patrick95470, 2016-09-21:
Ok, thx I’ll do like that

regards

patrick

markwal, 2016-09-24:
I think the last few messages indicate that this was a problem with using the wrong slicing profile with the CuraEngine plugin.  Maybe.  It looks like it is resolved?  If not, feel free to reopen and add more info @patrick95470.

closed 2016-09-24 by markwal

patrick95470, 2016-09-25:
Hi

Yes You can close it

kind regards