OctoPrint/OctoPrint #1505: switch to protocol relative and path relative includes

pcnate, 2016-09-21:
Please switch to relative paths for resources. This helps to allow compatability with webserver proxies.

protocols appear to be hard coded to http

    packed_libs.js?35d69079:formatted:3502Mixed Content: The page at 'https://octoprint.example.com/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://octoprint.example.com/api/files/local/powerAdapterMount.gcode'. This request has been blocked; the content must be served over HTTPS.

Assets are loaded with absolute paths

    <link href="/static/webassets/packed_libs.css?6c7d6974" rel="stylesheet" media="screen">

Javascript also points directly 

    var BASEURL = "/";
    var API_BASEURL = BASEURL + "api/";

These all make it difficult to use a proxy and force https. 

Switching to protocol relative paths and urls would fix the issue:

    //example.com/resource.any

Actually there is no need to specify the domain unless loading cross-domain, so:

    /resource.any

vs

    resource.any

foosel, 2016-09-21:
All URLs OctoPrint hands out depend on the incoming request, which you need to tell your proxy to set up properly. If you configure your reverse proxy to set `X-Script-Name` and `X-Scheme` headers as described [in the docs](http://docs.octoprint.org/en/master/configuration/config_yaml.html#server), all those URLs should be correct. There are also examples [here on the wiki](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples).

1.2.16 will also support the full set of headers `X-Forwarded-Host`, `X-Forwarded-Proto`, `X-Forwarded-Server` and `X-Forwarded-Port`.

You'll need to set those headers for things to work correctly, since OctoPrint's APIs - which the web interface utilizes - are meant for third party consumption and generate `Location` headers and other URLs into resource links contained in the JSON bodies, which need to be fully qualified in order to be easily consumable by any kind of potential client. Making the URLs inside the HTML of the web interface itself path and protocol relative would not be a solution to the general problem and still produce issues for you.

tldr: Please configure your reverse proxy correctly.

pcnate, 2016-09-21:
My apologies for opening an issue in this case.

Thank you for the links, I'm sure this will help me get it configured correctly.

On a side note, I searched the docs for "proxy" and found nothing. Needing to know what to search for is the most important thing...

closed 2016-09-21 by pcnate