OctoPrint/OctoPrint #1508: Hotend target temp not updated until temp is reached

mrvanes, 2016-09-23:
When starting a print, the hotend is instructed to heat up to certain temp (which happens) but Hotend target temp is still off, both in graph as in setting table. The hotend graph also reflects the belated target temp, just after temp was reached. Bed temp target is updated immediately.

markwal, 2016-09-24:
We need to see a bit from the terminal window showing what your printer firmware is saying about the temperatures while you're waiting for it to reach temperature.  Please copy and paste the contents of the terminal window.  Be sure to include the gcode command that is causing the wait as well as all subsequent responses.

pouncingiguana, 2016-09-29:
Further information on this bug as I've experienced it:

It appears that the hotend temperature is not updated (in the octoprint web interface or the API) when the printer is in the "Operational" state. This is noticeable when I manually set the hotend temperature, or when a print it finished (The temperature stays at what it last reported while printing). Even when polling the API it will continue to read the same temperature until the printer state changes (start printing, etc) or until I send another command to set the temperature. After sending the command it will update the temp 1 time, but will not update it again until another command is sent or the printer state changes.

I'm running Octoprint 1.2.16 and have verified this behavior on a SeeMeCNC Eris and Rostock Max v3.

foosel, 2016-09-29:
@pouncingiguana educated guess: your printer is running repetier firmware and you don't have all checkboxes tagged "Repetier" checked under Settings > Features (yes, I know, these need to be moved)

mrvanes, 2016-10-03:
My printer is a Lulzbot Mini and runs "Marlin" as far as I understand it. I'd love to share more info (terminal window) but it's quite verbose so please tell me what you want to see?

foosel, 2016-10-03:
@mrvanes all of it basically, but most importantly the parts including the heatup command right to when it starts updating the temperature graph again. What @markwal said. Verbosity is not a problem, too little information however is

mrvanes, 2016-10-03:
This is one of those embarrasing moments that you try to demonstrate a problem and it's no longer there. I've seen this since I started using Octoprint, even with latest version 1.2.16, but I can't reproduce tonight. Oh well... close the bug and I'll report back when I can, or leave it open for @pouncingiguana to report terminal output?

foosel, 2016-10-04:
Huh, that's odd.

In general, an issue like you were encountering can happen in two cases when entering a blocking heatup (`M109` or `M190`): the firmware doesn't report back the temperature while heating up (it normally should send a message every second or so with the current temperature) or it does report that but the format is weird making OctoPrint either not recognize that it's looking at a temperature report or unable to parse out the current temperature. My guess is either the first or second happened here, which is why both @markwal and me were asking for a terminal log, since it would clarify which case of both we are looking at.

That it's now working all of a sudden however makes me wonder a bit if there might be a third option I'm not aware of, because both options mentioned above shouldn't just go away on their own without any changes to either the firmware or OctoPrint.

About the "same" issue reported by @pouncingiguana, it's not, they stated that they didn't get temperature updates why the printer was in state "Operational", which indicates that it doesn't send the `M105` request to the printer regularly. That it will only do while in that state if it thinks a blocking heatup or other blocking command is already active. Repetier firmware reports the `ok` and temperature report on separate lines, with the `ok` preceding the temperature line, making the temperature line look like it was triggered by a blocking heatup started from within the firmware itself (e.g. via a command from the LCD or an SD file) instead of just being a response to a regular run of the mill `M105`. Which is why that external heatup detection needs to be disabled for Repetier firmware or exactly this kind of behaviour will happen right on the very first `M105` response received by OctoPrint. And AFAIK SeeMeCNC ships Repetier firmware on their printers, hence the educated guess.

Long story short: since you can't reproduce it anymore right now and the "me too" by @pouncingiguana was a red herring, personally I'm not too keen on keeping this on my active TODO list until it shows itself again (in which case you could always request a reopen of this ticket). But I'll leave the final decision to you and just mark the ticket as unreproduced for now.

mrvanes, 2016-10-05:
Just for the sake of closure, this is the first part of my terminal output after starting a print:

Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 G26*28
Recv: ok
Send: N2 G21*24
Recv: ok
Send: N3 G90*19
Recv: ok
Send: N4 M82*29
Recv: ok
Send: N5 M107*32
Recv: ok
Send: N6 G92 E0*65
Recv: ok
Send: N7 M140 S60*84
Recv: ok
Send: N8 G28*27
Recv: X:0.00 Y:191.00 Z:159.00 E:0.00 Count X: 0 Y:19196 Z:254400
Recv: ok
Send: N9 M109 R140*101
Recv:  T:199.6 /140.0 B:20.9 /60.0 B@:127 @:48 W:?
...

foosel, 2016-10-14:
> ```
> Send: N9 M109 R140*101
> Recv:  T:199.6 /140.0 B:20.9 /60.0 B@:127 @:48 W:?
> ```

My guess is it's because OctoPrint doesn't know about the `R` parameter yet (it only parses out the `S` parameter so far and to be quite honest I didn't even know about that parameter until just now - naturally the reprap wiki and the Marlin source code disagree again how it's used), hence won't extract the target temperature from the command and that combined with #1543 then led to the target temperature in the frontend not updating until new temperature data was received from the firmware. In your example above that happens pretty much immediately, but the question is if that was also the case earlier. We will probably never know considering you can't reproduce it anymore.

mrvanes, 2016-10-14:
I must admit I have also updated the firmware for the printer since got it and although I didn't suspect coincidence, it might have... we'll never know and as for the problem: it has completely disappeared for me.

ScottWell1, 2016-12-20:
I can confirm this behavior was a problem with early Lulzbot provided Marlin 1.00 releases.  It is not an Octoprint issue.

In the Lulzbot/Marlin 1.0.0 code, the M109 with Rxxx (i.e., wait for bed temp to reach xxx) was blocking and did not include bed temperature in it's periodic status response.  Thus, any software (Octoprint, Cura, Simplify3d, etc.) did not see any bed temperature update until the target was reached and M109 completed, thus allowing an M105 to process. 

This problem was fixed in the Lulzbot/Marlin 1.0.2 release.  That release added bed temperature to the periodic status messages issued during the blocking "M109 Rxxx" command, and Octoprint is able to monitor (and update in GUI) the current bed temperature during the blocking M109.   The subsequent 1.0.8 release, and the current 1.1.10 release, also include bed temperature in the M109 output.

Suggest closing this as "not an Octoprint issue" (it was a Lulzbot/Marlin 1.0.0 issue).

foosel, 2017-01-09:
@ScottWell1 the issue @mrvanes described related to the hotend temperature not being updated though - was that also an issue with Lulzbot Marlin 1.00? Are you aware of any tickets relating to this issue I could take a look at?

In any case though, errors in reporting temperature information are usually not OctoPrint related but caused by firmware issues (no responses or wrong format on those reponses). So since the issue is no longer showing itself for the original reported anyhow, I'll consider this closed.

closed 2017-01-09 by foosel

ScottWell1, 2017-01-09:
@foosel I need to read more carefully.  :-)   

I misinterpreted the issue reported here as being that the **actual temp** was not being updated during heatup.  That was a known issue in older Marlin, as the blocking extruder/bed heat commands did not include temperatures of BOTH bed and extruder in the periodic status messages.   That prevented Octoprint (and Cura/Pronterface, and Simplfy3d, etc.) from getting both temperatures when a blocking heatup was in process. 

Having re-read this original issue post, I now see it plainly referred to the "target temp" field rather than the "actual temp" field.  Firmware would presumably have no effect on that, since Octoprint doesn't need a response from firmware to obtain/display that value.  I promise to drink my entire cup of coffee before reading or responding on any other issues.  :-)

foosel, 2017-01-09:
> Firmware would presumably have no effect on that, since Octoprint doesn't need a response from 
> firmware to obtain/display that value

Ah, but in general the firmware has the final word :) If you send something like `M104 S400` through OctoPrint and the firmware instead sets a target temperature of say 220degC which it reports back on the next response to `M105`, that one wins. OctoPrint will take the target temperature value from a temperature command (as longs as it can parse that, the `R` parameter was still unknown in earlier versions) but only until it reads the next target value from the firmware.

> I promise to drink my entire cup of coffee before reading or responding on any other issues. :-)

I've seen people for whom that would have made things worse actually, so be careful with that coffee :)

ScottWell1, 2017-01-09:
@foosel Marlin 1.0.0 (at least as released by AO for the Lulzbot Mini) did not include target temperature (or any bed temp/target) in the "M109 Rxxx" responses.  The format of the 1sec blocking status was simply "T:[degHotEnd(tmp_extruder)]  E:[int(tmp_extruder)] W: ?".   That's it.  No target temp, no bed temp, and no support for multiple extruders.   Just current hotend temp and extruder number.

You commented on Oct-14, saying:
> "_OctoPrint doesn't know about the R parameter yet (it only parses out the S parameter so far_".

That, combined with the lack of Target temperature in the blocking status messages, I think provides explanation for the behavior @mrvanes reported when he was running the original firmware:  Octoprint didn't parse the "Rxxx" to get target temperature, nor did the blocking status messages have a target temperature for Octoprint to harvest, so Octoprint left the target temp field in the GUI at "Off".

Marlin 1.0.2 and later (as released by AO for Lulzbot Mini) expanded the blocking status messages to include both Current and Target temps, for both bed and extruder(s), as well as information about extruder and bed wattages.  (In other words, the format is essentially identical to the M105 response, but without the leading "ok".)   

Once @mrvanes updated to 1.02 or later, the blocking responses included Target temperature.  Octoprint was then able to get Target temperature from the blocking status messages, and update the GUI accordingly -- even though it did not parse the target from "M109 Rxxx". 

So I *think* we have both an explanation for the behavior seen, and an explanation for why it was resolved with newer firmware.

**The remaining detail is whether you think parsing the "Rxxx" is worthwhile.**  At least with today's Marlin, and as long as Octoprint reads/updates target temperature from the blocking status message,  then the GUI will get updated correctly with target temperature roughly one second after the "M109 Rxxx" is sent (i.e., as soon as Octoprint parses the first blocking status message).

> I've seen people for whom that would have made things worse actually, so be careful with that coffee :)

Agreed, more caffeine might just help me make wrong conclusions at a faster pace.  :-)

foosel, 2017-01-10:
> but I think you already added this in later versions?

Yeah, it's supported since 1.2.18.

ringerc, 2019-03-02:
I'm seeing something that matches this (also with Marlin - ADVI3++) on 1.3.10. I'll investigate further and if if I can reliably reproduce it I'll open an issue. But interestingly I see the same issue when working without octoprint, directly with the printer LCD - both on factory firmware and ADVI3++. What's fascinating is that the target temp is reported from the printer as 0 to both the control LCD and Octoprint, and whether Octoprint or the control LCD initiates the job.