OctoPrint/OctoPrint #1509: hot end just shy of set point

mbiancolo, 2016-09-24:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

mbiancolo, 2016-09-24:
Just updates newest           and now hot end gets to about 243      set point is  253    floats up and down  a little   and know it can do more because light  turns off and on     is there a setting for time or deviation 
That maybe got changed in the new version    and its turning the heater off thinking its going to overshoot     Thanks   Mike

foosel, 2016-09-24:
This is in fact answered in the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things

In short: OctoPrint is not responsible for controlling your printer's heaters, it just tells your printer to go to your selected temperature and it's your printer's job to hear up to that.

Also, next time please actually follow the instructions for creating a ticket, thank you.

closed 2016-09-24 by foosel

mbiancolo, 2016-09-24:
Really worked Fine   before the update !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

foosel, 2016-09-24:
Then please fill out the bug template *completely* and then I'll happily reopen the ticket of its clear that it is indeed OctoPrint fault. Be sure to include the contents of your terminal tab, because that will clearly show what OctoPrint is sending to your printer when you send a temperature command, and what your firmware is doing with that information. Make sure you have read the linked FAQ entry too.

As things are now, you sadly have provided not even slightly enough information to be able to debug your problem of in fact it is a problem with OctoPrint. I can't solve problems I know nothing about but "it's broken, fix pls".

foosel, 2016-09-24:
Still missing the terminal tab contents up there showing the problem. Again, that's necessary in order to be able to observe the issue. Happy to wait until your pla print is finished but the terminal tab contents showing the communication between OctoPrint and the printer at the time of the issue (temperature set and not reached) are essential.

Also please give me a quick ping here via a comment once you've added that info, I don't get a notification from github when you edit an existing issue.

mbiancolo, 2016-09-24:
added link   really stumped  here is also 
https://gist.github.com/mbiancolo/88d7884db95a1213e28eaccf2aea6e69
dont know if you need me to to do fancy log will try and get that as well

mbiancolo, 2016-09-24:
added fancy log   just doing a ramp up to temp
https://gist.github.com/mbiancolo/e100645d54b61abf407ea317bd683788

nophead, 2016-09-24:
Looks like you need to tune your extruder heater's PID constants. When it
gets close to target it reduces power too much and the temperature falls
again. It is oscillating.

It's nothing to do with OctoPrint. The target temperature is 253 and it
never gets there.

foosel, 2016-09-25:
Yep, exactly what @nophead said, and what I said from the very beginning and what is stated in the linked FAQ entry. 

OctoPrint tells your printer "go to 253". You printer doesn't. Your printer also wouldn't with a former OctoPrint version or when you'd sent it the command manually via the serial line. That needs to fixed in you printer. Performing a PID tuning is a good step to solve this.