OctoPrint/OctoPrint #1511: Setup on a Raspberry Pi running Raspbian [System Commands] Errors

krwynn, 2016-09-25:
Hi,

I'm getting errors after performing this section. 
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#system-commands

Everything else when installing OctoPrint on my Raspberry Pi 3 went smoothly until this section. The system tab is ok and is as expected. However when trying to reboot or shutdown I get these messages:

Reboot:
Command failed with return code 1: >>> /etc/sudoers.d/octoprint-shutdown: syntax error near line 2 <<<
>>> /etc/sudoers.d/octoprint-shutdown: syntax error near line 2 <<<
sudo: parse error in /etc/sudoers.d/octoprint-shutdown near line 2
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin

Shutdown:
Command failed with return code 1: >>> /etc/sudoers.d/octoprint-shutdown: syntax error near line 2 <<<
>>> /etc/sudoers.d/octoprint-shutdown: syntax error near line 2 <<<
sudo: parse error in /etc/sudoers.d/octoprint-shutdown near line 2
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin

Any help would be appreciated.

Thanks

nophead, 2016-09-25:
Looks like the file /etc/sudoers.d/octoprint-shutdown is corrupt. It should
only have one line:

pi ALL=NOPASSWD: /sbin/shutdown

krwynn, 2016-09-25:
Thanks. I have no idea where this should be. I thought I was in the right place. I apologize.

krwynn, 2016-09-25:
Thanks. I thought thus was in the right place, but evidently not. Very confusing. I apologize.

foosel, 2016-09-26:
Instructions used to say `^D` to tell you "please press Ctrl+D here" (which is a common way to type that out in the linux world). Some people (including you) appear to have copied that straightly into the sudoers file however which is admittedly an easy mistake to make. Setup guide has been adjusted to prevent this from happening again.

closed 2016-09-26 by foosel

krwynn, 2016-09-26:
That makes perfect sense and is exactly what I did. Thanks so much for clearing that up. Appreciate it!