OctoPrint/OctoPrint #1512: Performance request - NOT an issue

sanchosk, 2016-09-26:
Hi. After installation of 1.2.16 the performance of the whole OctoPrint is much better.
I am running installation on slow Orange Pi ONE (512MB RAM only).
The system spends 20% of resources just on webcam streaming alone (crappy onboard camera interface + crappy module with driver).
Before 1.2.16 when I accessed OctoPrint, the ongoing print would froze for 2-3 moves and then continue.
Now, the print runs smoothly, no delay or empty cache detected at all and the page loads on 2 devices in parallel no problem!
Congratulation to the improvement. I'll keep track should the performance go down or something.

foosel, 2016-09-26:
@sanchosk Weird - because I didn't change anything in 1.2.16 that should have that much of a perceivable impact O_O Sure that wasn't something that changed on your end after all? 

Anyhow, closing this since apparently all is great at the moment and as you said yourself it's not an issue ;)

Comments are still possible though :)

closed 2016-09-26 by foosel

sanchosk, 2016-09-26:
I changed nothing. The same setup works as before. I even printed the same thing as before (small drawer).
I'll keep track of this thing and possibly update this "issue" :)

foosel, 2016-09-26:
@Eriobis there have been no changes that should have made the client "heavier" between 1.2.12 and today. Most changes in the frontend where cosmetic or small bug fixes. The only thing I could think of which would made initially startup of the client feel a bit slower is that I removed the cause of a race condition in 1.2.13, but there's no way around that small delay in startup and it shouldn't cause a general slowdown, only stuff taking a bit longer on initial connect. In general I spend a lot of time in development not only making the server leaner but also the client, so if at all it should become faster over time. If it doesn't, something is wrong. The UI feeling "sluggish" is usually a sign for network trouble though - requests not going through fast enough, the push service being throttled etc.

In order to make sure it really is caused by the update(s) and not something else, I'd ask you to please [roll back](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image) to 1.2.12 (or whichever was the version before you started noticing your problem) and see if the problem goes away. If it does go away, please open a full bug report, including the filled out ticket template plus a list of your installed plugins and your config.yaml in a gist or pastebin link (to check your configured size thresholds and other UI settings that might influence performance on mobile - please remove the entries ``accessControl.salt`` and ``server.secretKey``). If the problem persists even after a roll back it might be worth looking into your installed plugins, any other changes that might have happened in the past 3 months (new webcam? different placement of the router?) and - if your server is currently connected via wifi - to check a wired connection to rule out any issues with your wifi.

Eriobis, 2016-09-26:
Sure, I didn't have much time for investigating into that problem, but I will take a closer look at this problem. I have touch UI plugin, but as I already said it used to be responsive and now it's sluggish. The raspi are wired and my network is solid. I don't see any network issue.
I will do some tests and give you a feedback and open an issue.

Thank you for all your good work !