OctoPrint/OctoPrint #1513: [Request] Option to allow sending of M0/M1 to the printer

foosel, 2016-09-26:
When encountering an `M0` or `M1`, OctoPrint currently will not send it to the printer but instead just pause locally.

It would be nice to be able to configure this behaviour and have OctoPrint send the commands after all, in case of additional "pause" functionality provided by the firmware, and at the peril of breaking stuff if those additional functionalities are implemented poorly and without having host communications in mind.

See #1448

hackermagnet, 2016-12-29:
The M0/M1 commands work fine from gcode files but they don't work using commands from custom buttons.
Is there any way to fix this?

foosel, 2017-01-09:
As mentioned above:

> OctoPrint currently will not send it to the printer but instead just pause locally.

So a custom button with an `M0` or `M1` will basically just act as a pause button right now (and do nothing when not printing or already paused). I don't know if that's what you mean with "don't work" vs what the same commands when streamed from a gcode file through OctoPrint will currently do.

yeltrow, 2017-07-02:
I like to use the manual console to tweak first layer height by doing the following:
Pause job.
M92 Z0.35; (if printing at 0.3 and I want to move down 50um)
G1 F50000 Z0.3; (move the print head down 50um)
Continue job.  

Enabling G1 commands via the terminal send would be a huge help to me.  Mabe there is another way?

foosel, 2017-07-03:
@yeltrow Uhm... `G0`/`G1` can be sent just fine, this ticket concerns `M0`/`M1` which are currently not sent to the printer but instead interpreted as local pause command. Am I missing something here?

CraigMason, 2017-12-29:
Quick suggestion, could we print something in the terminal if M0 is triggered? I spent a while today diagnosing why M0 was pausing, but not showing anything on the LCD (including re-flashing firmware twice). And the terminal just ignores the command with no output.

Thank you!

emphasize, 2018-01-12:
> So a custom button with an M0 or M1 will basically just act as a pause button right now (and do nothing when not printing or already paused).

Does it? Trying to setup custom controls here to perform a "manual" bed probe.
Like that:
G1 X10 Y10
M0 Click to resume
G0 X10 Y190
M0 Click to resume
G0 X190 Y190
M0 Click to resume
G0 X190 Y10

The printer performs the path, but doesnt pause and wait for "click to resume". Not as an array of commands, not as a single custom command button and not through the terminal.

fiveangle, 2018-01-12:
>> Without ULTIPANEL M0 does nothing.

> I have one (SmartLCD) but does Octoprint "know" that?

The note should probably be fixed to replace "_ULTIPANEL_ ..." with "_a printer control panel with display configured,_ ...". But it was written at a time where most Marlin users understood that "ULTIPANEL" was synonymous with "a generic control panel with display".

The note is warning that M0 does nothing if no panel is configured because if Marlin did execute the M0 w/o a panel, it would pause the printer awaiting user input without any way for user to enter said input.  
 You have a panel so the `M0` command should execute as expected.

According to Gina, Octoprint currently does not pass M0 so Marlin will never see it and never pause, even if it supports it.  Sending M117 just displays text on the panel, which is expected to work, since M117 is not blocked by Octoprint.  If you send M0 with some other host software such as Repitier, you should see the expected pause behavior in Marlin.  If you don't, it means it's broken and you should file an issue at the Marlin github tracker.

-=dave

foosel, 2018-05-05:
[image]

Added a new option to change behaviour. This will be checked by default (current behaviour). When OctoPrint blocks `M0` or `M1` it will also now log a corresponding message to the Terminal tab and `serial.log`.

Pushed to `maintenance`, soon `devel`, to be released with 1.3.9

closed 2018-07-25 by foosel

hackermagnet, 2019-08-21:
I'm pretty confused with these options.
I want to pause with M0 and resume by pressing the button on my 3d printer's lcd.
This works fine when I print directly from my 3d printer's sd card.

So I need to send the M0 command to the firmware and pause octoprint at the same time. mWhen I press the button I want to resume octoprint.
Now, when I send M0 my_text, I see my_text on the screen and octoprint pauses AFTER I press the button on the printer. Then I need to resume also from octoprint.
How could I make octoprint to pause before pressing the lcd's button and resume after pressing it?

commit 81471df4 (2018-05-05): Option to send M0/M1 to the printer

Also added a message to the serial output if M0/M1 is blocked.

Closes #1513