OctoPrint/OctoPrint #1516: ATX power by temperature

NickRimmer, 2016-09-26:
HI, I think it will be helpful.

It's about ATX power switch via GPIO. Before switch off, hot end must be cooled. So how we can send via GPIO (or execute shell script) after print will be complete and temperature lower then X?

Main question or request in executing command after the end of print with waiting some temperature.

Thank you (;

foosel, 2016-09-28:
I'm not sure I understand what you want correctly, so please correct me if I'm getting this wrong. From what I understand you are looking for a way to have the ATX power supply for your printer shut off via a GPIO pin once a print finishes, but with a delay so that the hotend fan can still cool down the hotend and prevent a plug from forming.

This should already be possible via [events](http://docs.octoprint.org/en/master/events/index.html), the ``PrintDone`` event specifically, using the linux [at command](https://linux.die.net/man/1/at) (which you can install on OctoPi with ``sudo apt-get install --no-install-recommends at``) to delay the command triggering the GPIO pin (e.g. ``echo "gpio write 4 low" | at now + 5 minutes`` to set GPIO pin 4 to low after 5 minutes). 

A more sophisticated approach could be done with a plugin which would also detect if a new print has been started and monitor the current temperature for a threshold value after which to shut things off.

NickRimmer, 2016-09-28:
Thank you for answer (=
You understood correctly. 5 minutes waiting is a solution, but not very precise. There is no relation to real temperature.

Ideal solution, is made customization scripts. Where we can set some triggers and actions (presets or/and GCODE scripts). 

For example: 

1.  "_Create script_" (or task or something like)
2. Set trigger "_On end print_"
3. Set actions "_Send GCODE_" with value "_M106 S255_" + "_M104 S0_" + "_M140 S0_" (there "+" I just mean on new line)
4. "_Wait temperature_" with value "_70_"
5. "_Execute shell script_" with value "_poweroff_atx.sh_"

This will perfect solution for most situations. Very helpful trigger can be like "After/Before print", "After start / before end super OctoPrint", "After time of idle", "On the time after print", "File name contain ..." or "File name ends with ...", "System time is ...", "System time between..." (it can be helpful if we want power off printer only at night or swith on/off light via GPIO) and some else. So triggers can used multiply and together with setting order.

Actions can be simple GCODE or Shell script or something what can't be executed by printer, but can by super OctoPrint server (= Also we can set in action, execute other script with own triggers and actions.

That will be awesome tool!

P'S: sorry for my english, I hope you understand me (;

MoonshineSG, 2016-10-19:
I'm  running Marlin and with a tiny change in the firmware, this can be done elegantly.

Enable `STAT_LED_RED_PIN` (if can be set to a fictional value - but I actually put an red LED on one of the spare pins to indicate "hot" printer), add 

```
        if (! new_led ) {
          SERIAL_PROTOCOLLN("//action:cooled");
        }
``` 
after 
```
        WRITE(STAT_LED_RED_PIN, new_led ? HIGH : LOW);
        WRITE(STAT_LED_BLUE_PIN, new_led ? LOW : HIGH);
```
in function `handle_status_leds` in `Marlin_main.cpp`

Then in OctoPrint have a plugin hook to `octoprint.comm.protocol.action` which can do whatever you want when the printer cools down to the value set in Marlin (default 55C)

Salandora, 2016-10-19:
This is one feature I wanna implement in my [Automatic On/Off Plugin](https://github.com/Salandora/OctoPrint-AutomaticOnOffPlugin) I'm currently developing.
It is forked from foosel's awesome PbPiLink Plugin.

If you use a Pi you'll need a [second Plugin](https://github.com/Salandora/OctoPrint-RaspberryPiApi) which provides the API for the first Plugin, in the settings you can easily select the pin to switch, for now it should be possible to use the no client timeout at something like 5 minutes to get what you desire.

If you want to use a Shell script well you'll need [this Plugin](https://github.com/Salandora/OctoPrint-CommandApi) :-)

All 3 Plugins are **not** listed in the Plugin Manager yet. So you need to install by URL.

For the "On end print" part well in settings you can find Gcode Scripts where you can define the end gcode.

Hope this will fit your needs. And as said the temperature part is one thing on my todo list for the main plugin.

gege2b, 2016-10-19:
Hi all

I think this feature should be host-independent

You can add some Gcode in your slicer's end gcode field : 

; at the very end
M109 S40 ; to set the temperature at a reasonable low value, printer should wait before proceeding further commands
M104 S0
M81 ; power off (on most firmware)

and voilà (this snippet would certainly needs some tweaking)

Of course, it's only possible if the firmware supports the "power off" command, and the motherboard have a dedicated/free pin to manage the /PS_ON signal on the PSU

foosel, 2016-10-19:
> ```
> M109 S40 ; to set the temperature at a reasonable low value, printer should wait before proceeding 
> ``` 

As I recently discovered, at least on Marlin `M109 S<temp>` will usually only wait for the temperature if the target is *higher* than the current - you can configure that behaviour, but it defaults to being this way AFAIK. If you want it to also wait for cooling, you'll need to use `R` instead of `S`: `M109 R40`

foosel, 2017-01-27:
The [PSU Control](http://plugins.octoprint.org/plugins/psucontrol/) plugin that was just released should take care of that.

closed 2017-01-27 by foosel