OctoPrint/OctoPrint #1517: Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.

knifefish, 2016-09-27:
#### What were you doing?

Updated to newest version of octoprint on raspberry pi 3. Can no longer connect to printer.

#### What did you expect to happen? 

Everything to work.

#### What happened instead?

Cannot connect to printer. Prusa i3 mk2

#### Branch & Commit or Version of OctoPrint

1.2.16

#### Printer model & used firmware incl. version

Prusa I3 Mk2 - Fimware 3.0.8

#### Browser and Version of Browser, Operating System running Browser

Chrome - 53.0.2785.116 m (64-bit) - Windows 10 64bit

#### Link to octoprint.log

<script src="https://gist.github.com/knifefish/b79477672db39e0455ec93c347a71148.js"></script>

#### Link to contents of terminal tab or serial.log

<script src="https://gist.github.com/knifefish/936be79e30e86baed74413e2e66d3a92.js"></script>

knifefish, 2016-09-27:
Octoprint.log - https://gist.github.com/knifefish/b79477672db39e0455ec93c347a71148
Serial.log - https://gist.github.com/knifefish/936be79e30e86baed74413e2e66d3a92

foosel, 2016-09-27:
```
2016-09-27 00:37:32,194 - SERIAL - DEBUG - Trying baudrate: 250000
2016-09-27 00:37:32,195 - SERIAL - DEBUG - Send: N0 M110 N0*125
2016-09-27 00:37:44,607 - SERIAL - DEBUG - Recv: ~��~��f�����������������~ �fx~fx���fx ��x �� ������x��x � f~`����~ �������f� xf�xf~��`������xf�~f������� f���fx���fx ��x ����f������`fxf�fx����f�~ �x��f�����f����f~x�xf����f~f���~ ��x�f���������x��`fx``f�xf~��`��fx`~����~�����xfx��f��������~f��fx������`�~��f�������fx �� �x�� ��`� ���f�������ff���ff� xffx`��f~f���������f��f��������x �x �f� �x�f� �~�x�� ��x � f�������ff���ff� � ����fx��f��f~������ ��������x���f��������x ��f��xf�f ��f������� ����fx��f��f~���������f������fx��f��f~�������� f��fx`��f��~ ������fx��f��f~��������f��������x  ������ f��� f������� xxf��fx`~��f����f���~ ��f�f��� xffx`��f~f�������f��� f��f��� ~f�~f� xffx`��f~f�������f��� �f��f�������ff� ��fx~f�f��`~����f���f��� f���ff���ff� fx�fx���������f��� �f��ff���ff� ��fx���������f��� �f��ff���ff� ��fx���������f���f��������x ����� f � ���� �x>�� ����x��ff�����������f��x�xfx`��������f��������x  f� �x� �� f������f` ��fx`~����~���~��f��������x ���`���x ����`��� f�������fx`��f��~���f�f��~~�� ������� `���fxfx ����������� ��~ ����x~ ���������f��������x � ���x�� `� � �� f�������fx��xfx�~ ��f�~f�f��~~�� ������� `���fxfx �������������f��������x ��� ``� f������� f~`����f�~f��~���f~�� x��f���f� ��~�� ����`fx��fx`��~f~`fx�������������xfx�����fx`��f��~���fx��xfxf�f���f��������x ����f������`����f�f��`��fx`~����~���~ `��f���fx��f��������` ��f��x ������
2016-09-27 00:37:54,628 - SERIAL - DEBUG - Baudrate test retry: 4
```

Well, apparently 250000 baud is the wrong one, and for whatever reason *the printer* doesn't reply anymore to consecutive attempts at lower baudrates, that makes auto detection fail. Might not always do it, which is why you are attributing that behaviour to the update, but from the looks of it it's really not (and OctoPrint's baudrate auto detection mechanism also hasn't changed in years).

What happens when instead of `AUTO` you manually select `115200` and THEN try to connect?

PS: Please don't remove any lines from the template, or the bot will complain like it did here.

knifefish, 2016-09-28:
It throws the same error regardless of what baud I choose manually.

foosel, 2016-09-28:
Did you restart the printer (power off and on again) before trying 115200 baud? Since in the other ticket you also mentioned you upgraded *OctoPi*, which version of OctoPi (you can find out with `cat /etc/octopi_version`) are you running and did you actually upgrade that too (as in, flashed a new image) or was that wrongly phrased in the other ticket? With which versions of both *OctoPrint* and *OctoPi* could you still connect to your printer? Does [rolling back](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image) OctoPrint to that version restore connectivity?

Also please try connecting from a different host to rule out any issues with your printer itself. Printrun/Pronterface for example (your MK2's manual contains instructions on how to do that)

knifefish, 2016-09-28:
Updated serial.log with 115200 baud. Power has been cycled multiple time. I misspoke when I said i upgraded octopi all I did was the prompted updated through the Octoprint web ui. Current version of Octopi 0.13.0. Will test roll back and different host when I get a spare moment.

https://gist.github.com/knifefish/af1f681e3a86a182e61e88c8837c6f08

knifefish, 2016-09-29:
Rolling back to 1.2.15 does not have an affect. Was able to connect with pronterface.

foosel, 2016-09-29:
The connection attempt with 115200 looks at least like it hit the right baud rate, but the firmware never sends an `ok` for either of the two `M110` that were sent, so OctoPrint assumes it's not talking to an actual printer (since that WOULD respond with an `ok`).

Please check if you have "Wait for `start` on connect" checked under Settings > Features. If not, check it, save, try to connect with 115200 baud again. Usually that is not necessary to be checked (and it causes issues with printers that do not reset on connect, which is why it's not checked by default), but some controllers need it or they tend to run into the issue you are observing now: The printer doesn't answer according to the protocol ("some command" - "ok") and so the initial connection handshake runs into a timeout.

ctgreybeard, 2016-09-29:
I have the same printer and firmware and I have absolutely no problem connecting to octoprint. I was wondering if the OP has other serial-USB devices connected to the Pi? Perhaps octoprint is trying to talk to a some device other than the printer? What does `ls /dev/tty*` show?

foosel, 2016-10-04:
@ctgreybeard it's the correct port he's connecting to, but the printer doesn't reply with an `ok` as it's expected to at the end of the initial handshake. Hence the suggestion to make OctoPrint wait a bit before trying to talk to it with the "wait for `start` on connect" checkbox.

foosel, 2016-11-08:
No reply from OP, considering this solved.

closed 2016-11-08 by foosel

EspenShampoo, 2017-08-30:
Hello, i have the same issues as OP. I have no idea what im doing wrong and it annoys me that everyone else is having no issue with octoprint and their raspberry pis while i just go from one obstacle to another. Im not great at computers but i do know a thing or two, this whole makes me feal like a dumbass -.-

here is my octoprint log:

https://gist.github.com/EspenShampoo/5723c0b9321cc9df4b477cf1adbc63b3#file-octoprint-log