OctoPrint/OctoPrint #1519: Could not parse output from pip

adammhaile, 2016-09-28:
#### What were you doing?

Trying to install plugins. This has failed for all I have tried thus far including:
STL Viewer, TouchUI, and Full-Featured Slicer

#### What did you expect to happen?

Install to succeed and plugin be available in settings.

#### What happened instead?

Could not parse output from pip.

[image]

This is despite the fact that the install log shows success:

Installing plugin "STL Viewer" from https://github.com/jneilliii/Octoprint-STLViewer/archive/master.zip...
/home/pi/oprint/bin/pip install https://github.com/jneilliii/Octoprint-STLViewer/archive/master.zip
Collecting https://github.com/jneilliii/Octoprint-STLViewer/archive/master.zip
Downloading https://github.com/jneilliii/Octoprint-STLViewer/archive/master.zip (116kB)
Requirement already satisfied (use --upgrade to upgrade): STL-Viewer==0.3.1 from https://github.com/jneilliii/Octoprint-STLViewer/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied (use --upgrade to upgrade): OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg (from STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): tornado<4.1,>=4.0.1 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): sockjs-tornado<1.1,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.1-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.6-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): requests<2.8,>=2.7.0 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Jinja2>=2.4 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): certifi in /home/pi/oprint/lib/python2.7/site-packages/certifi-2016.02.28-py2.7.egg (from tornado<4.1,>=4.0.1->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): backports.ssl_match_hostname in /home/pi/oprint/lib/python2.7/site-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado<4.1,>=4.0.1->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): blinker in /home/pi/oprint/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Babel>=1.0 in /home/pi/oprint/lib/python2.7/site-packages/Babel-2.3.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): speaklater>=1.2 in /home/pi/oprint/lib/python2.7/site-packages/speaklater-1.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): webassets>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/webassets-0.11.1-py2.7.egg (from Flask-Assets<0.11,>=0.10->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): argh>=0.24.1 in /home/pi/oprint/lib/python2.7/site-packages/argh-0.26.1-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): pathtools>=0.1.1 in /home/pi/oprint/lib/python2.7/site-packages/pathtools-0.1.2-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): pyasn1>=0.1.3 in /home/pi/oprint/lib/python2.7/site-packages (from rsa<3.3,>=3.2->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): regex in /home/pi/oprint/lib/python2.7/site-packages/regex-2016.04.15-py2.7-linux-armv7l.egg (from awesome-slugify<1.7,>=1.6.5->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): Unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages/Unidecode-0.04.19-py2.7.egg (from awesome-slugify<1.7,>=1.6.5->OctoPrint->STL-Viewer==0.3.1)
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages/MarkupSafe-0.23-py2.7-linux-armv7l.egg (from Jinja2>=2.4->flask<0.11,>=0.9->OctoPrint->STL-Viewer==0.3.1)
[... cut ...]
packed_app.js?354484a8:10190 ... dependency resolution done
packed_app.js?354484a8:10580 Initial application setup done, connecting to server...
packed_app.js?354484a8:9170 Connected to the server
packed_app.js?354484a8:10567 Finalizing application startup
packed_app.js?354484a8:10472 Going to bind 26 view models...
packed_app.js?354484a8:10540 ... binding done
packed_app.js?354484a8:10559 Application startup complete

#### Screenshot(s) showing the problem:

See above

I have read the FAQ. 
Yes

ss89, 2016-10-03:
Have you tried to just restart your octoprint instance?
I am currently facing the same issue, but for plugins that are already installed (but still have the install button in the plugin manager).
The Slicer Plugin installed just fine yesterday (new instance spawned).
It seems that octoprint isn't aware of the output /Requirement already satisfied \(use --upgrade to upgrade\):.*/ but i haven't looked in the code, so foosel will know better. ;)

foosel, 2016-10-04:
If a plugin is already registered with the system, OctoPrint will disable the install button. So I'm unsure right now why the plugin wasn't detected as being registered when you tried to reinstall it again, maybe I'm not properly detecting present-but-disabled plugins here...

From the log provided by @adammhaile it also looks like the plugins get installed just fine (they show up one after another in the freshly generated plugin list after an install albeit marked as disabled since they all require a restart of OctoPrint to become active) and a restart would directly activate them just fine. OctoPrint should on initial install also have shown a corresponding message informing about the necessary restart.

From the looks of it, I would currently guess that the screenshot in fact was generated on attempted reinstall of an already installed but disabled plugin, so just what @ss89 is reporting - could this be the case?

As a side note, the last change to the plugin manager happened in May with 1.2.11, and that was merely adding support for the About dialog. The actual plugin installation code hasn't been changed since May of 2015. So if it still worked with 1.2.15 and stopped working with 1.2.16, it's either something that changed in the environment (What pip version is that? You can find that information in the plugin manager below the list of plugins that are currently installed), something that changed with regards to the involved steps or - if it is related to a reinstall like suggested above - simply a bug that has been present for ages but was never reported since it usually doesn't trigger when people just install plugins once.

foosel, 2016-10-05:
@adammhaile need some input, see comment above :)

adammhaile, 2016-10-05:
So... the details I logged were in fact from the second or third time I had
tried to install the plugins. The first time, it gave me the same error but
sadly I don't have those logs (or am not sure where to find them). After if
happened the first time I rebooted the system. After it happened the second
time, I also rebooted and the plugins where still not showing up. But the
power went out last night and I'm looking now and, sure enough, everything
is there. It's been several days now since I tried installing them
*and* rebooted.
So not sure what's up.

foosel, 2016-10-05:
At least I found the reason why the already installed plugins where not detected as installed in the plugin repository browser. The repository file contained the wrong value for the plugin id (a path instead of only the key). My current guess is that this was caused by a Jekyll update on Github Pages (where the repository is hosted) a couple months ago. I have fixed it now which should at least make plugins that are properly detected as installed by the plugin manager actually be filtered/marked as installed in the repository browser.

That should also make reinstalls work, since reinstalls do two passes, one of which forces an actual reinstall via ``--force-reinstall`` which makes the reinstall actually execute even if it's the same version and OctoPrint report success.

@adammhaile about your (former) problem now

> and the plugins where still not showing up

Just to make sure, where did you look? The plugins you tried to install were not visible on any of the pages (with 5 plugins per page) under the plugin manager itself after a restart of OctoPrint?

[image]

Or were they visible there but their UI components were not active?

Because according to your shared `octoprint.log` they at least were detected by the system itself after each install just fine, albeit disabled since no restart had taken place at least according to the log yet... 

Before install:

```
2016-09-27 14:31:08,509 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.5.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
```

After first install (Printer Stats):

```
2016-09-27 14:43:57,685 - octoprint.plugin.core - INFO - 8 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.5.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
| !Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
```

second install (STL Viewer):

```
2016-09-27 14:45:01,346 - octoprint.plugin.core - INFO - 9 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.5.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
| !Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/softwareupdate
| !STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
```

and finally third install (TouchUI):

```
2016-09-27 14:46:09,785 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.5.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/pluginmanager
[... cut ...]
| !STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
| !TouchUI (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.16-py2.7.egg/octoprint/plugins/virtual_printer
```

And if they were detected by the system, visible in the plugin manager (which uses the same data as those log segments) but their UI components were not there, that sounds like more of a caching issue than a plugin installation issue.

foosel, 2017-01-27:
@adammhaile I am actually still waiting for input from you here ;) Or is this no longer relevant? If so please say so so that it can be closed.

adammhaile, 2017-01-27:
Ah, sorry about that. The plugins were getting installed and I think just
the first time around it failed to show success, but then was showing
failure when I tried a second time. But once I rebooted they plugins were
available. Guess I didn't know I had to reboot first.
All is fine now :)

foosel, 2017-01-27:
Ok, I'm considering this closed then :)

closed 2017-01-27 by foosel