OctoPrint/OctoPrint #1520: [Request] Drag, Drop, and Go!

sniglet, 2016-09-28:
Once a printer (and Pi, and Octoprint) are all well tuned, it would be nice to just have a hopper/bin/chute to throw files at with the understanding that the owner just wants the file to be printed. 

An option to alter the "Drag and Drop" landing fields in the UI to 'Drag, Drop, and Print' might be useful...with the caveat that it's up to the end user to make sure the last print was removed from the build plate and they'll make sure they observe that a spaghetti monster isn't printing due to bad adhesion.

ntoff, 2016-10-06:
I have a batch file that I can just right click on a gcode file and go to the send to menu (in windows) and click send to printer and it'll upload and print. Far quicker to right click a file than open a browser window.  Also having to deliberately use a different method of uploading takes care of that responsibility thing you mentioned since I'm not uploading via octoprint's GUI but rather via the command line.

sniglet, 2016-10-07:
Care to post it? I'm on a Mac, but I might be able to produce an OS X equivalent.

ntoff, 2016-10-07:
OSX you could install curl https://curl.haxx.se/download.html

This is to upload to octoprint direcly from Simplify3d [ taken from here](http://community.robo3d.com/index.php?threads/simplify3d-octoprint-gcode-direct-to-octoprint-setup.8217/) 

curl -k -H "X-Api-Key: yourkeyhere" -F "select=true" -F "print=true" -F "file=@[output_filepath]" "http://octopi.local/api/files/local"

and maybe just replace @[output_filepath] which I think is s3d's variable with whatever osx's command line input variable would be, is it ${1} ? I forget. Note: I've never actually used the curl version so couldn't say for sure.

And for windows, this is what's in my batch file:

C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command "$wc = new-object System.Net.WebClient; $wc.UploadFile('http://octopi.local/api/files/local?apikey=APIKEYHERE&select=true&print=true', '%1')"

eyal0, 2017-04-13:
This is a duplicate of #272

foosel, 2017-04-18:
@eyal0 indeed, at least in a way... #272 asks for a third upload button, but I'd say drag-drop-n-go fairly falls within the intention behind that original ticket. So let's merge the two, aka close this one & add a note to #272 - further discussion in #272 please :)

closed 2017-04-18 by foosel