OctoPrint/OctoPrint #1525: [Request] Add temperature logger

christianlupus, 2016-09-29:
From time to time it might be interesting, which temperature the hotend/bed actually had for a given time.

This can be interesting for firmware enhancements (optimization of controllers) but also in debugging of printer issues (e.g. did the filament break because the temp was too low and therefore could not pass the extruder?). Ideally some layer information (Z value) might get handy.

The data should ideally be written to a file on the octoprint server in a text format or something similar for further processing (CSV). This would also allow to log temperatures while not been logged in the web interface.

markwal, 2016-09-30:
This could easily be done via a plugin actually.

foosel, 2016-10-01:
A plugin I might add which - if everything works out - I'll be writing
anyhow for a customer in the near future and which will be released as open
source then :)

christianlupus, 2016-10-04:
That would be perfectly fitting. I am looking forward for it.
Thank you for the great piece of software so far.

foosel, 2017-01-27:
Things went a little differently than originally planned and while said plugin does exist, it also turned into a all-in-one solution for various very specific requirements. So I don't think it's the solution to this particular problem (and also not yet published). So this is still up for grabs.

tedder, 2018-05-31:
This should be closed, dupe of #321 (I think).

foosel, 2018-05-31:
@tedder #321 is about adding additional sensor values to the embedded temperature graph, this one here is about creating a CSV file or something similar with various data points.

ErikAmlie, 2022-11-02:
Did this become a plugin in the end? Could be really usefull for my project

ErikAmlie, 2022-11-02:
Did this become a plugin in the end? I coudl really need a CSV of time, XYZ and temperatures for my layer adhesion master thesis

cp2004, 2022-11-02:
There is no plugin I know of - it would be very easy to create one with a few lines of Python. There is a hook that can be called when the temperatures are received from the printer that can be used to log it.

Billiboy71, 2023-02-23:
@foosel 
is there already a plugin for logging the temperatures for the truck order?