OctoPrint/OctoPrint #1527: Motor off not working as expected

christianlupus, 2016-09-30:
#### What were you doing?

I just surfed the web frontend and tried to disable the steppers (to move the head manually).

#### What did you expect to happen?

The steppers are free to move and the head is "force-free".

#### What happened instead?

The printer did not stop pumping current through the steppers resulting in high holding torque generated.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.16 (master branch): 354042b

#### Printer model & used firmware incl. version

Prusa i3 with stock firmware (Repetier 0.91 according to terminal). I have nothing done with it yet. So where can I get more information about this?

#### Link to octoprint.log

http://paste.debian.net/848596/

#### Link to contents of terminal tab or serial.log

http://paste.debian.net/848595/

I have read the FAQ.

#### Potential solution / workaround ####

The octoprinter sends a `M18` command to disable the steppers. However my firmware seems to ignore this setting. But the code from Slic3r disabled the steppers correctly. They use command `M84` instead

markwal, 2016-09-30:
You can add your own control that sends M84, if you like.  You can either edit config.yaml directly to do that, or add the custom control plugin to allow you to add them using the mouse (right click) in the browser: http://plugins.octoprint.org/plugins/customControl/

foosel, 2016-10-01:
According to http://reprap.org/wiki/G-code M18 disables all steppers. M84
stops the idle hold. Another case of one firmware doing one thing and
another doing another thing and there being two options in the first place
which of course can't happily coexist. It's frustrating.

Long term solution should be to allow to configure the command sent by the
built in commands (at least the things that are not simple movements), or
maybe even move them into prefigured custom controls or something.

Short term solution would be to use a tiny plugin which rewrites the M18 to
an M84 on sending. Would basically be a slightly adjusted version of
http://docs.octoprint.org/en/master/plugins/hooks.html#rewrite-m107-py-https-github-com-octoprint-plugin-examples-blob-master-rewrite-m107-py,
just changing the rewrite routine and getting rid of the logging.

christianlupus, 2016-10-04:
Yes, I think the rewrite rule will work. I was not aware of this feature.
However you might want to include the device in the compatibility list to support in the long time run the printer. Thank you very much.

closed 2016-10-04 by christianlupus