OctoPrint/OctoPrint #1528: Discoeasy and Octoprint

charettemtb, 2016-10-03:
Hello,

I have a Discoeasy200 (from Dagoma) and I'm not able to use Octoprint with it.
Octorpint doesn't consider it connected but the communication seem's active.
I'm using the last version of Octoprint.

Here is the logs:

Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0xaf7e1bb0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110 N0*125
Recv: echo:Marlin 1.1.0-RC6
Recv: echo: Last Updated: 2016-04-24 12:00 | Author: Dagoma.Fr
Recv: Compiled: Sep  2 2016
Recv: echo: Free Memory: 3545  PlannerBufferBytes: 1232
Recv: echo:V23 stored settings retrieved (396 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z2560.00 E98.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z4.00 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X9000 Y3000 Z100 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P65.14 I12.50 D84.84 C100.00 L20
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.60
Recv: echo:SD card ok
Recv: OK

Thanks

foosel, 2016-10-03:
Is it really

    Recv: OK

at the end? Not this?

    Recv: ok

If it's really uppercase it's a firmware bug, it should be lower case according to the protocol.

charettemtb, 2016-10-05:
Yes it is really in uppercase.
I transfer this information to Dagoma.
Does I have the ability to adapt the software to use this syntax?
Thanks

foosel, 2016-10-10:
> Does I have the ability to adapt the software to use this syntax?

I'm not sure if you are asking whether you can easily modify OctoPrint or the firmware to fix this issue.

The change in OctoPrint would be basically this [patch](https://en.wikipedia.org/wiki/Patch_(Unix)) (untested!):

``` patch
diff --git a/src/octoprint/util/comm.py b/src/octoprint/util/comm.py
index cf4b5d5..88bcc39 100644
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -966,8 +966,10 @@ class MachineCom(object):
 						except ValueError:
 							pass
 
+				lower_line = line.lower()
+
 				#If we are waiting for an M109 or M190 then measure the time we lost during heatup, so we can remove that time from our printing time estimate.
-				if 'ok' in line and self._heatupWaitStartTime:
+				if 'ok' in lower_line and self._heatupWaitStartTime:
 					self._heatupWaitTimeLost = self._heatupWaitTimeLost + (time.time() - self._heatupWaitStartTime)
 					self._heatupWaitStartTime = None
 					self._heating = False
@@ -1021,6 +1023,7 @@ class MachineCom(object):
 					self._changeState(self.STATE_PRINTING)
 					self._clear_to_send.set()
 					line = "ok"
+					lower_line = "ok"
 				elif 'Done printing file' in line and self.isSdPrinting():
 					# printer is reporting file finished printing
 					self._sdFilePos = 0
@@ -1039,14 +1042,14 @@ class MachineCom(object):
 							pass
 				elif 'Done saving file' in line:
 					self.refreshSdFiles()
-				elif 'File deleted' in line and line.strip().endswith("ok"):
+				elif 'File deleted' in line and lower_line.strip().endswith("ok"):
 					# buggy Marlin version that doesn't send a proper \r after the "File deleted" statement, fixed in
 					# current versions
 					self._clear_to_send.set()
 
 				##~~ Message handling
 				elif line.strip() != '' \
-						and line.strip() != 'ok' and not line.startswith("wait") \
+						and lower_line != 'ok' and not line.startswith("wait") \
 						and not line.startswith('Resend:') \
 						and line != 'echo:Unknown command:""\n' \
 						and self.isOperational():
@@ -1099,7 +1102,7 @@ class MachineCom(object):
 							self._errorValue = "No more baudrates to test, and no suitable baudrate found."
 							self._changeState(self.STATE_ERROR)
 							eventManager().fire(Events.ERROR, {"error": self.getErrorString()})
-					elif 'start' in line or 'ok' in line:
+					elif 'start' in line or 'ok' in lower_line:
 						self._onConnected()
 						self._clear_to_send.set()
 
@@ -1109,14 +1112,14 @@ class MachineCom(object):
 						startSeen = True
 						self._sendCommand("M110")
 						self._clear_to_send.set()
-					elif "ok" in line:
+					elif "ok" in lower_line:
 						self._onConnected()
 					elif time.time() > self._timeout:
 						self.close()
 
 				### Operational
 				elif self._state == self.STATE_OPERATIONAL or self._state == self.STATE_PAUSED:
-					if "ok" in line:
+					if "ok" in lower_line:
 						# if we still have commands to process, process them
 						if self._resendSwallowNextOk:
 							self._resendSwallowNextOk = False
@@ -1139,7 +1142,7 @@ class MachineCom(object):
 						else:
 							self._logger.debug("Ran into a communication timeout, but a command known to be a long runner is currently active")
 
-					if "ok" in line or (supportWait and "wait" in line):
+					if "ok" in lower_line or (supportWait and "wait" in line):
 						# a wait while printing means our printer's buffer ran out, probably due to some ok getting
 						# swallowed, so we treat it the same as an ok here teo take up communication again
 						if self._resendSwallowNextOk:
@@ -1154,7 +1157,7 @@ class MachineCom(object):
 							elif not self.isSdPrinting():
 								self._sendNext()
 
-					elif line.lower().startswith("resend") or line.lower().startswith("rs"):
+					elif lower_line.startswith("resend") or lower_line.startswith("rs"):
 						self._handleResendRequest(line)
 			except:
 				self._logger.exception("Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:")
```

However if you are unfamiliar with changing things in OctoPrint I'd advise against this, especially since you might run into trouble when updating (either losing the change again or producing conflicts).

The far better (and probably also easier) fix would be to adjust the firmware. My guess is that the simply somewhere put `OK` as acknowledgement message instead of `ok`. So you are looking at probably only a two character change and a quick firmware recompile. 

Note that your printer vendor is obligated by the open source firmware's GPL license to provide you with the sources to its modified version of it - ask them for it. You fix such things yourself if they don't do it in a timely manner.

Meanwhile I'm closing this ticket since it's a firmware bug. While OctoPrint can be modified to work around it, a line has to be drawn somewhere.

closed 2016-10-10 by foosel