OctoPrint/OctoPrint #1529: simple reset to Octoprint for new user

thomllama, 2016-10-04:
OK my brother sent me a Pi and installed octopi.. everything works but, he signed into Octoprint and can't remember the user/password... is there a way to just reset octoprint without starting from scratch? (which I tried with new SD card but for some reason the pi won't read it)

donovan6000, 2016-10-04:
Delete `~/.octoprint/users.yaml` to remove all existing users, and change `firstRun: false` to `firstRun: true` in `~/.octoprint/users.yaml` to make OctoPrint think that it's being run for the first time. Then restart OctoPrint and it'll display the dialog box for creating a new user.

thomllama, 2016-10-04:
Ok,, but where/how is this yaml?  Looked on the card directly, not there,  tried through SSH, no file or directory is all I get back.   I'm completely clueless with coding..

donovan6000, 2016-10-04:
The files should be located at `~/.octoprint/users.yaml` and `~/.octoprint/users.yaml`.

I've never used OctoPi before, but, based on quickly looking though its [repo](https://github.com/guysoft/OctoPi/tree/devel/src/filesystem/home/pi/.octoprint), running the following will perform all the tasks that I mentioned.
```
rm ~/.octoprint/users.yaml
sed -i -e 's/firstRun: false/firstRun: true/g' ~/.octoprint/config.yaml
/etc/init.d/octoprint restart
```

foosel, 2016-10-04:
Small correction: 

```
rm ~/.octoprint/users.yaml
sed -i -e 's/firstRun: false/firstRun: true/g' ~/.octoprint/config.yaml
sudo service octoprint restart
```

thomllama, 2016-10-04:
thanks guys.. will try.. guessing I can just copy paste this into the terminal once logged into the pi?

thomllama, 2016-10-04:
Hey Donovan, you still doing the M3Dfio stuff?  man you were a life saver wit that POS!! LOL  only way to get that thing to do anything decent !!

thomllama, 2016-10-04:
THAT WORKED!!!! thanks guys!!!  not a friggin clue what that all was but it worked!!!  _(right now I'm like the 1990's internet dancing banana gif !! )_

foosel, 2016-10-04:
[image]

(SCNR)

I guess it's time to add this to the FAQ finally, been meaning to for a while now...

thomllama, 2016-10-04:
ya!!! that banana dude!!!!  :)

foosel, 2016-10-04:
This is now also part of [a new FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#i-forgot-my-octoprint-password-how-can-i-reset-it). Closing.

closed 2016-10-04 by foosel