OctoPrint/OctoPrint #1534: Serial.log file handling

sanchosk, 2016-10-06:
#### What were you doing?
When enabling the serial.log communication, it works OK (I did not noticed any slow-down as the warning mentions and I am on a pretty slow HW, btw), the file is not divided per day.
That means that after 2-3 days of occasional prints, the file has 390MB.

#### What did you expect to happen?
I expected the log file to be separated in similar fashion as the octoprint.log file - per day. I assume it would be difficult to split the file in mid-print, so I would consider to name it according the date when the print started - e.g. the serial.log is continuously updated as it goes (logging e.g. temperatures from the printer) and then, when I hit Print button, it do something like this:
- is the serial.log creation date today? If yes - keep it and start printing
- If not, move it to serial-YYYY-MM-DD.log (similar to octoprint.log) and start a new serial.log

#### What happened instead?
I have one big continuous serial.log file growing significantly within a week.

#### Branch & Commit or Version of OctoPrint
Hmmm, I just updated to 17RC1, I think this was 16RC2, but I'm not sure :( But all of the latest releases have been doing serial logging the same way.

#### Printer model & used firmware incl. version
Reprap i3 clone with Repetier Firmware.

#### Browser and Version of Browser, Operating System running Browser
Firefox on Linux + Windows

#### Link to octoprint.log
Not helpful in this issue.

#### Link to contents of terminal tab or serial.log
The file is 390MB big, I'd say it's not worth sharing :(
It's easily replicable.

#### Link to contents of Javascript console in the browser
Not relevant.

#### Screenshot(s) showing the problem:
Not relevant.

I have read the FAQ.
Yes :)

foosel, 2016-10-10:
While you shouldn't be leaving this on (it really does impact performance negatively), you are right, something is off there. It should be restricted to 2MB in size via Python's own logging subsystem, but apparently that's not working as expected. I'll need to take another look there. Thanks for reporting!

foosel, 2016-10-13:
This has been changed now in such a way that the `serial.log` will be rolled over when connecting to the printer. The backup copies will get their last modification date attached. This does NOT solve the issue of very large `serial.log` files when staying connected over days or weeks, but that should not be done with `serial.log` enabled in any case - it is really only meant to be a debugging tool, to be enabled when required to debug a communication issue, and it is *not* to be left on during regular operation outside of troubleshooting due to its performance impact and also the fact that it simply produces way too much data for normal operation which is only useful for troubleshooting anyhow.

The changed behavior has been pushed to `maintenance` (and soon `devel`) and will be part of the upcoming 1.2.17 release (since that sadly requires another RC anyway).

closed 2016-10-13 by foosel