OctoPrint/OctoPrint #1535: [Request] Resize Terminal Window?

gcormier, 2016-10-07:
Would be nice to be able resize the terminal window within a browser.

Runolas, 2016-11-30:
What do you mean by resize the terminal window ? You want to adapt the terminal size depending on the browser size, or to resize it manually ?

gcormier, 2016-11-30:
Yup, the window you enter commands and see the output. Either/or would be awesome!

Runolas, 2016-12-07:
I found in my virtual environment directory, a css file named "octoprint.css". I can edit it to set the minimum height of the terminal, identified by the id "terminal-output" (default is 340 pixel). This change takes effect after restart octoprint server. However, when I connect my printer, the terminal size reset to the default value. I think it's due to the autoscroll of terminal. I tried to set the pixel maximal height of the autoscroll (id "pre-scrollable"), but it doesn't work.

Runolas, 2016-12-07:
We can set manually the terminal size in the browser developer tools, but it's need to set it for each connection !

JayceeB1, 2020-05-25:
Up! need this :)

jneilliii, 2020-05-25:
Sounds like a plugin idea to me.

jneilliii, 2020-05-25:
A quick test in the developer console, the following line seems to re-size properly so setting these css rules in Themeify may actually work.
```
$('#term,#terminal-output').css({'width':'800px','height':'800px','max-height':'800px'});
```

jneilliii, 2020-05-25:
Yep, seems to work in Themeify....here are the settings.

| selector         | css-rule   | value |
|------------------|------------|-------|
| #terminal-output | max-height | 800px |
| #term            | height     | 800px |
| #term            | width      | 800px |

And here is the output....

[image]

JayceeB1, 2020-05-26:
Wow, nice job!
Thanks a lot!

corbolais, 2020-06-29:
Would be nice to have it in octoprint core. Or at least a hint on the terminal page that if you want to change settings/size you need to install the `themeify` plugin.

cp2004, 2021-05-18:
This works in several plugins that can add custom CSS, and I remember @LazeMSS saying he might add it as an option from the UI customizer plugin 😉.

closed 2021-05-18 by cp2004

LazeMSS, 2021-05-19:
> This works in several plugins that can add custom CSS, and I remember @LazeMSS saying he might add it as an option from the UI customizer plugin 😉.

Yes is already implemented in next version 🤗