OctoPrint/OctoPrint #1537: DisplayXYZ instead of DisplayZ

sim0njo, 2016-10-07:
Hi,

OctoPrint was such a great improvement for using my 3D printer that it brought me to the idea of using the printer for more than just printing.

I mounted a dial-micro-meter onto my printer's printhead and using the 'Control' tab of OctoPrint I can move it around in XYZ axes, to measure relevant coordinates of a part placed on the printbed.

But what I miss is a display of the current XYZ position, either automatically updated or with a separate 'Update' button.

Another usefull option would be a 'Tag' button that when pressed, appends the current XYZ coordinates to a scrollable editbox, so it is easy to copy/paste a group of coordinates.

So what I am looking for is an extension of the ControlZ plugin -> ControlXYZ plugin.

thx in advance
Jo

sim0njo, 2016-10-07:
[image]

foosel, 2016-10-07:
You can define a custom control button that when pressed will show you the current x, y and z coordinates. It's even post of the example in the documentation for custom controls, see [here](http://docs.octoprint.org/en/master/features/custom_controls.html) :)

Having it automatically update as part of the state display however is a bad idea, why is [explained in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#why-doesnt-octoprint-show-me-the-current-hotend-position-anywhere).

sim0njo, 2016-10-07:
Tried it and it works, problem solved

Thx for the fast response.

foosel, 2016-10-07:
Perfect, closing then :D

closed 2016-10-07 by foosel