OctoPrint/OctoPrint #1539: [Request] Plugin: status.html

Loup911, 2016-10-08:
The plugin create a file status.html on the web server. 

The fils contain some detail of the print in text.

*****
Status: printing
File: alpha.gcode
Print time: 01:22:33
Print time left: 02:00:30
Current height: 6.95mm
Current layer: 216/543
progression: 50%
Hothend: 241/240C
Bed: 59/60C
*****

the file is automatically generated based on a pre-established time limits.

in the plugin option it would be nice to change the time limits of the generation of the file and you can select the data to be displayed there.

Booli, 2016-10-18:
Hey @Loup911 ,

it seems like you want a solution to a problem you have and want to solve it with this status.html page. The solution proposed is strange in the context of OctoPrint especially if you are familiar with all the features OctoPrint already offers. 

One thing that immediately comes to mind is the great REST API that OctoPrint offers, supplying nice JSON data for almost all information you want. 

Have look at the (great) documentation on the API: 
http://docs.octoprint.org/en/devel/api/general.html

The current job info: 
http://docs.octoprint.org/en/devel/api/job.html#get--api-job

The printer status: 
http://docs.octoprint.org/en/devel/api/printer.html#get--api-printer 

Those two combined will give all the info you want in nice clean JSON, which is easily parsed and can be polled by yourself on your own frequency. 

Again, I don't know exactly what you goal is, so maybe a total different approach is even more sensible(like a plugin). Please elaborate on what you are trying to solve with the status.html idea.

Booli

foosel, 2016-11-08:
Goal is still unclear, requested functionality as far as goal can be understood is already available. Closing this since there's apparently no interest by OP to explain the reason behind this request further.

closed 2016-11-08 by foosel