OctoPrint/OctoPrint #1540: [Brainstorming] using ramdrive during print

Loup911, 2016-10-09:
I wondered if it would be possible at the start raspberry creates a ramdrive and make a copy of the script in the octoprint ramdrive.

Octoprint run from ramdrive. All config files are on sd-card.
When printing is start print file is temporarily copied to the ramdrive for printing. At the end of printing the file is deleted from ramdrive. The printing data are write on the SD card.

The accessibility of the file on a ramdrive is much faster than the SD card.

It remains to know how much RAM the operating system and software are needed. And how much space would be needed for the ramdrive to contain the script octoprint and one Gcode file.

FHeilmann, 2016-10-11:
G-Code files can grow to any size depending on the model. This can't be done in a "fire and forget" manner since if the file is too big, you'd have to fall back to the normal behavior.

Using a fast SD-card (or even a USB Thumb drive) bundled with a strong serial connection should be fine for most models. Moreover, you can copy each G-Code file to a printer-attached SD card to eliminate the serial or OctoPi related bottlenecks.

foosel, 2016-10-14:
I'm not entirely sure what that's supposed to achieve. During printing, the SD from which the file is read is not the bottle neck. The serial communication to the printer is. And as @FHeilmann pointed out, GCODE files can go into the hundreds of MB, making a RAM disk on something as RAM restricted as a Pi pretty much useless.

So I'm not entirely sure what the added overhead of moving the file over to another drive implemented as RAM disk prior to printing is supposed to achieve - could you elaborate @Loup911?

Loup911, 2016-10-16:
Improve the speed of access to the file. reduce the number of read and write on the SD card to avoid destroying some sector on the sd card.

On the first generation of raspberry it contained 250 mb which was divided into two. 128mb for the CPU and 128mb for the GPU

As octopie work terminal instance and not using the GUI, the GPU as needed 16mo of RAM. So there have 224 MB of RAM available. It is true that on this version it would be difficult to transfer the script and gcode file into RAM.

I think for the first generation it would be possible to do for only the script octoprint and web server file.

Depending on the raspberry generation of the amount of memory for ramdrive could be increasing, to allow to place the gcode being printed. If the file is larger than the ram space allocated in function raspberrry generation is printed the print take the file from SD card.

script for the RBP revision 
http://www.raspberrypi-spy.co.uk/2012/09/getting-your-raspberry-pi-revision-number-using-python/

Revision list
http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/

foosel, 2016-10-16:
Again, the speed of the card is irrelevant here. It is not the limiting factor in transferring the data to the printer, the by several orders of magnitude slower protocol conform communication on the serial interface is.

Reading doesn't reduce the lifetime of the sd card, and you'd still have to write the file at least once to disk, with or without a ram drive as intermediary (whose speed - again - would be completely irrelevant). 

If you don't believe me: Try uploading a file to the printer's SD card via OctoPrint. The first part of that process will be uploading the file to the pi, which will be done in a fraction of the time needed to transfer it to the sd card right after (press of upload button to streaming start vs streaming start to end). And that is comparing file writing to streaming - try downloading the file via the little download button in the file list, it will be even faster. As a friendly advice: don't do this  test with a large gcode file, keep it smaller than 50KB or you'll be waiting a long time for stuff to finish streaming. That "sending data to the printer" part, THAT is your bottleneck. Not the reading or even the writing speed of the SD.

I'm sorry, but apart from a way higher process complexity, cross platform implementation and maintainability nightmares and a way higher potential for stuff to go wrong and ruin prints as a consequence of that, I still don't see anything that would add. And no positive effects at all.

closed 2016-10-16 by foosel

nophead, 2016-10-16:
Actually reading does wear out NAND flash, which is what SDcards are made
from but nowhere near as fast as writing. A read reduces the strength of
the data so it has to be rewritten eventually. the controller does this
transparently and moves the data about to wear it evenly.

foosel, 2016-10-16:
Considering that the writes a regular running Linux system will perform on said sd card by far outweigh any reads or writes performed on a single gcode file stored in OctoPrint, and that the gcode file also would need to be read when it would need to be read I think we can safely ignore this minor implementation detail, especially in the context of the general topic of this discussion, don't you think?

nophead, 2016-10-16:
Yes considering how many times I have printed the same files over and over
again I haven't noticed any SD cards wear out.