OctoPrint/OctoPrint #1541: change default profile via API

MoonshineSG, 2016-10-10:
#### What were you doing?

change the default printer profile via the API 

curl --insecure --connect-timeout 15 --request PATCH -H "Content-Type: application/json" -H "X-Api-Key: XXXXX" --data '{ "profile": { "default": true } }' https://octoprint.local/api/printerprofiles/_marlin

#### What did you expect to happen?

the new profile to be set as default.

#### What happened instead?

the new profile is shown as default both in Settings and via the api (https://octoprint.local/api/printerprofiles), but an OctoPrint restart "forgets" about it and resets the default profile to the old one

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0.dev1417+g73e8d5e (devel branch)

#### Printer model & used firmware incl. version

not important 

#### Browser and Version of Browser, Operating System running Browser

Chrome & API via CURL

#### Link to octoprint.log

n/a

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s) showing the problem:

n/a

I have read the FAQ.

MoonshineSG, 2016-10-10:
At the same time a related question. If I have 2 profiles with custom ID, why is there a "_default" named profile as well (I did not create that) ?

MoonshineSG, 2016-10-11:
there seams to be a problem with the "colour" as well. The profile colour does not get applied to the UI (top bar)

ntoff, 2016-10-14:
>there seams to be a problem with the "colour" as well. The profile colour does not get applied to the UI (top bar)

It never has for me either, always wondered what that colour thing was meant to be for.

MoonshineSG, 2016-10-14:
I found a work around to all these problems. Instead of having multiple profiles and do an API switch (which doesn't work), I have a single default profile and using 2 API calls I change the properties of that profile (extruder count and nozzleDiameter) and with the settings API I change the colour and the printer name to show the count and the diameter. 

A third, custom plugin call will reload the UI to reflect the changes. 

Works perfectly.

foosel, 2016-10-14:
> It never has for me either, always wondered what that colour thing was meant to be for.

Best don't ask. An external requirement from my former employer for more or less internal use that (color coded printer profiles in another UI and me not arguing against it loud enough). In fact, I should probably just kill it again.

foosel, 2016-10-14:
About the original problem, that has now been solved on `maintenance` and `devel`. While the new default value was set on the settings object, it wasn't saved to disk. Which means while OctoPrint was running, it persisted, but once restarted it fetched the old value from disk.

> If I have 2 profiles with custom ID, why is there a "_default" named profile as well (I did not create that) ?

Because the `_default` one currently has to be there always due to how stuff is designed at the moment. I might change that in the future so that when creating your printer profile (which devel will now actually offer you a wizard for if you haven't customized it) it will also allow you to specify the id instead of having the first profile always be `_default`, but for now, it's simply like that due to an admittedly bad design decision.

MoonshineSG, 2016-10-19:
I can live with the current `_default` :D 
Thanks for the other fix.

commit 3f8dde75 (2016-10-14): Persist settings after changing default printer profile

Fixes #1541
closed 2016-10-19 by MoonshineSG