OctoPrint/OctoPrint #1544: [bug report] Login issues on 1.2.17rc1 with IE11 and Android Browser

avoid200, 2016-10-12:
#### What were you doing?
I have updated from version 1.2.16 to 1.2.17rc1.

#### What were you doing?
I have done an update via the web interface.
After that I try to log back into Octoprint.

#### What did you expect to happen?
The login should have worked.

#### What happened instead?
It works only with Firefox (49.0.1), not with IE (11.0.9600.18349) and also not on Android Smartphone / Tablet. Only Firefox reacts when the login button is pressed, the other browser does not.

#### Branch & Commit or version of OctoPrint
Before the update 1.2.16.
Now 1.2.17rc1.

#### Printer model & used firmware incl. Version
Printer: Vellemann Vertex K8400 (electronics replaced by Ramps 1.4).
Firmware: Marlin 1.1.0-RC7 - 31 July 2016

#### Browser and Version of Browser, Operating System running Browser
Windows 7 Pro 64Bit / Android
Firefox (49.0.1) / Internet Explorer (11.0.9600.18349) / Android standard browser

#### Link to octoprint.log
https://gist.github.com/anonymous/13012510ec5e4f9af25b531ed187e94d#file-octoprint-log

#### link to contents of terminal tab or serial.log
unavailable.

#### link to contents of javascript console in the browser
unavailable.

#### Screenshot (s) showing the problem:
unavailable.

ntoff, 2016-10-13:
Works fine here on Windows 10 Pro 
Octoprint: Version: 1.2.17rc1 (rc/maintenance branch)
 IE 11.494.10586.0
"Internet" on Android 6.0.1 (whatever browser that is) on a Samsung tablet.

Have you tried clearing your internet cache?

foosel, 2016-10-13:
Just tested on IE myself: 11.633.10586.0. Login worked perfectly fine. Don't have an Android device on hand ancient enough that it still doesn't ship with Chrome as browser by default, so couldn't cross check there personally. 

The only change between 1.2.16 and 1.2.17rc1 even remotely related to login behaviour was making sure to *on logout* also delete the old `remember_me` cookie, not only the new suffixed version `remember_me_P<port>`.

Which made me highly doubt an issue with OctoPrint here and rather something like a plugin interfering (by producing a js error that prevents the login from being properly processed). I noticed in your log (after splitting it back up into multiple lines :P) that you added the "Full-featured Slicer" plugin during updating to 1.2.17rc1 as well. So I just installed that and lo-and-behold, it contains Javascript code that can't be processed by IE11 and probably also not your Android tablet, the back ticks in [this](https://github.com/kennethjiang/OctoPrint-Slicer/blob/master/octoprint_slicer/static/js/slicer.js#L94) and [this](https://github.com/kennethjiang/OctoPrint-Slicer/blob/master/octoprint_slicer/static/js/slicer.js#L116) line to be precise (and possibly more places, that's only what I could find so far). I [reported](https://github.com/kennethjiang/OctoPrint-Slicer/issues/10) that in the plugin's bug tracker, please follow that ticket. Closing this one since it's not a bug with OctoPrint but a plugin that you installed.

BTW: The javascript console is a must to debug things like this, and it *is* available even in IE11 by hitting "F12" and then selecting "Console".

closed 2016-10-13 by foosel