OctoPrint/OctoPrint #1545: [Request] Include username of user who started print in "State" pane

twilde, 2016-10-14:
When access control is turned on, it would be really helpful to have the username of the user who started a print (or loaded the file for printing, or maybe both?) in the "State" pane, either all the time or as an option. Thanks for the awesome software!

somebugis, 2016-10-28:
Hello @twilde 
I'll try do to it this week

somebugis, 2016-11-02:
What do you mean by have the username ? Just the user who inquire his name before the print or rather a login page ?

twilde, 2016-11-02:
I was referring to the username the person logged in with, when access control is turned on.

somebugis, 2016-11-04:
[image]

This one ?

twilde, 2016-11-04:
Yup, that's the one!

cmtec, 2017-01-05:
+1 
When running a multi user environment this would help a lot.

foosel, 2018-06-19:
This is available on `maintenance` and `devel` and will be released as part of 1.3.9

closed 2018-07-25 by foosel