OctoPrint/OctoPrint #1546: Bug when printer is work and connect to http interface

userosos, 2016-10-16:
Hi everyone!
I has Raspberry pi 2, web-cam is connected to raspi, powersuplay has current 2 A. I downloaded an octopi from http://octoprint.org and i uses it. When my printer (reprap, corexy) in working and i go to in http interface my printer has some jerks (all motors is off ~1 sec after it all motors is on ~1 sec and it repeated ~10 sec). It happen all time when i connected to the web-interface of octoprint. In htop i can see that octoprint load CPU of raspi on ~150% in the moment after this bug i can see that octoprint use ~0% CPU.

Version: 1.2.16 (master branch) - octoprint

uname -a
Linux octopi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux

userosos, 2016-10-27:
I changed this ticket. Check it pls.

foosel, 2016-10-31:
Duplicate of #1241, please follow this ticket for progress on this and thinks to check.

closed 2016-10-31 by foosel