OctoPrint/OctoPrint #1547: The status is showing over 100%

Eriobis, 2016-10-18:
#### What were you doing?

Printing as usual

#### What did you expect to happen?

Get the status bar correctly 

#### What happened instead?

The status showed 100% ans the print wasn't finished
It finished at 125%

#### Branch & Commit or Version of OctoPrint

Version: 1.2.17rc4 

#### Printer model & used firmware incl. version

Prusa I3 MK2

#### Browser and Version of Browser, Operating System running Browser

Chrome Stable Build

#### Link to octoprint.log

http://pastebin.com/rvB0kRSh

#### Link to contents of terminal tab or serial.log

Not Relevant

#### Link to contents of Javascript console in the browser

If I only knew how to grab it :(

#### Screenshot(s) showing the problem:

[https://s13.postimg.org/q3msyvkd3/Octoprint_Problem.png](url)

I have read the FAQ.

foosel, 2016-10-19:
Note how it also says that it has printed 33.2MB of 29.9MB. OctoPrint depends on the information from the file system here. At the beginning of the print it asked the operating system "How big is that file" and got back "29.9MB". Then, over the course of printing the file while reading it line by line, it also asked the OS "where am I in the file" and it got back increasing numbers. 

For whatever reason, OctoPrint was able to read more bytes from the file than were originally supposed to be in it. So something is seriously wrong with your file system here, not OctoPrint, because that should never happen on the OS level if everything is working as expected.

The only other thing I could imagine happening here was that you somehow managed to modify the file externally after the print was already started but possible before the file was opened, that would be a windows of a couple µs only though.

I also notice that it says yet another size in the file list, it's claiming that the file is only 14.1MB there. So it definitely sounds like the file was being modified outside of OctoPrint by something or someone else, or that you should quickly back up your data because your file system is corrupt. What exactly did you do prior to starting that print?

Eriobis, 2016-10-19:
Hi, Thanks for all this pertinent information. I didn't take a look at the file size.
I didn't modify it while it was printing, if I did it was not on purpose.
I see the second file in the list is exactly the same name, I will investigate on this on my own.

On a second note, you mentionned filesystem. My new setup ( not completed yet ), is a USB drive  plugged into the pi and holding the data. The filesystem is FAT32, will it be compatible with the octoprint system ?
I'd like to avoid having all the files on the pi, as I found that sdcards aren't reliable enough. I would like to keep the filesystem and the software. All the other folders ( /tmp /run ) will be in a ramfs drive.

As for the "bug", you may close the request. As I mentionned, I will take a look at the file but I'm not sure I will be able to reproduce this.

closed 2016-10-19 by Eriobis

foosel, 2016-10-19:
FAT should be fine. OctoPrint is platform agnostic and doesn't really care about the exact file system as long as the OS is giving its the expected information about the files (e.g. what files are there, how are they named, what size are they and when were they last modified).

A word of advice however: If you found your SD cards to be unreliable, you should look into switching your SD brand, and also make sure to always shutdown your Pi AND wait for the LEDs to stop flickering before pulling the power. In nearly 4 years of printing from various Pis and accumulating files on them, I've never had anything get lost or corrupted. And the only issue I ever had with a corrupted SD (on one of my home servers) was my own fault, thoughtlessly pulling the power cord at the wrong time while it was in the middle of a Linux update.

Eriobis, 2016-10-25:
Thank you. I am changing my sd for some more reliable ones.