OctoPrint/OctoPrint #1551: Incorrect print size exceeds volume warning

ctgreybeard, 2016-10-20:
#### What were you doing?

Loading a gcode preparing to print

#### What did you expect to happen?

The file should load with no warning

#### What happened instead?

Received a warning that "Object doesn't fit print volume"

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0rc1 (rc/devel branch)

#### Printer model & used firmware incl. version

Original Prusa i3 MK2 (Firmware 3.0.8)

#### Browser and Version of Browser, Operating System running Browser

Chrome on iPad

#### Link to octoprint.log

http://pastebin.com/0p8cpRq5

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

Attached
[image]
[image]
[image]

My config.yaml: http://pastebin.com/K9a8GK3n

I have read the FAQ.

ctgreybeard, 2016-10-20:
Here's the gcode file that I was printing ...

[AJ_Taller_Cone_Tip.gcode.zip](https://github.com/foosel/OctoPrint/files/542917/AJ_Taller_Cone_Tip.gcode.zip)

JohnOCFII, 2016-10-26:
I've seen this message with the two prints I've tried since upgrading to Version: 1.3.0rc1 (rc/devel branch)

The actual print is quite a bit smaller than the print bed.  I was able to close the warning and print successfully.

[image]

[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/552274/octoprint.log.txt)

foosel, 2016-10-27:
The issue with this feature (and the reason why it's marked "Beta" under Settings > Features and can also be disabled there) is that the only way it can perform this check is by checking the object's bounding box (so min vs max coordinates of the printed model, determined by looking for moves which include extrusion) with the print volume's bounding box (so min vs max coordinates of that, as provided in the currently selected printer profile). For determining if an object fits on the bed, not only is the object's size relevant, but also its location on the bed (because that is determined by the sliced GCODE file).

In the screenshot by @JohnOCFII it's visible that the object apparently performs extrusion at Y=-3.0, which exceeds the print bed's smallest value for the Y axis of 0, hence the warning is produced. @ctgreybeard it would be great if you could check if that is the same for you.

My guess is that this is some kind of nozzle purging or something similar prior to the actual print start and hence perfectly valid. If that is indeed the case, I'll probably default to "off" for this feature for the stable release, since it might confuse people otherwise (it's really merely a warning, you can just ignore it and print stuff even if its coordinates exceed the bounding box of the print volume).

ctgreybeard, 2016-10-27:
Ah! I should have thought of that. Yes, the gcode for Original Prusa printers does move to a minus Y spot in order to do a wipe of the extruder.

The relevant lines are:

```
G1 Y-3.0 F1000.0 ; go outside printing area
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```

Rather than turning the option off I would propose that small minus excursions be allowed, perhaps on an additional option? I think this is a valuable feature and I would like to be able to use it on my printer.

Perhaps the printer minimums could default to 0 but be able to be specified as negative numbers? Clearly -Z is unlikely but -X or -Y is possible and even necessary in my case.

JohnOCFII, 2016-10-27:
> Rather than turning the option off I would propose that small minus excursions be allowed, perhaps on an additional option? I think this is a valuable feature and I would like to be able to use it [...]

Or would it make sense to separately define a "priming box" which could also be excluded from the bounding box check.  Users would be expected to populate the box.

foosel, 2016-11-09:
I've gone with the suggestion of @ctgreybeard and added the possibility to define a custom bounding box in the printer profile which exceeds the coordinates of the print volume, by manually specifying minimum and maximum X, Y and Z coordinates (limited to the print volume bounding box):

[image]

That should solve this.

If the checkmark is not set, everything is like before:

[image]

Will be in 1.3.0rc2

foosel, 2016-11-25:
1.3.0rc2 is out

closed 2016-11-25 by foosel

JohnOCFII, 2016-11-26:
Confirmed that the custom bounding box works correctly to avoid the warning for me in 1.3.0rc2.

ctgreybeard, 2016-11-26:
Same here. Thanks!