OctoPrint/OctoPrint #1552: Autoconnect does not work following printer error

rurwin, 2016-10-20:
#### What were you doing?

Reconnecting to the printer after an error in the print.

#### What did you expect to happen?

OctoPrint should connect just as it does following a reboot or restart

#### What happened instead?

OctoPrint reported that no serial port could be found.
'Error: Failed to autodetect serial port, please set it manually.'

#### Branch & Commit or Version of OctoPrint

1.2.16 (Master Branch)

#### Printer model & used firmware incl. version

SeeMeCNC Orion.
SeeMeCNC build of Repetier 0.92.2

#### Browser and Version of Browser, Operating System running Browser

Firefox 49.0
Linux Mint DE 17.1

#### Link to octoprint.log

http://pastebin.com/P33YqYdu

#### Link to contents of terminal tab or serial.log

Changing monitoring state from 'Error: C hit floor

' to 'Offline'
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Failed to enter programming mode'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2016-10-27:
How did you originally connect to your printer, also via autodetect or did you select your port manually then?

If the former I could imagine your printer's controller entered some mode of operation after the error that didn't allow entering programming mode, which the autodetect mechanism for the serial port to use attempts. In general I'd suggest to rather manually select the port manually when you know what it is for that reason.

rurwin, 2016-10-27:
Via autodetect.
I will try specifying it explicitly tonight.
However, does this explain why the situation is corrected by restarting OctoPrint, but not the Pi or the printer? Surely if the fault was in the printer then it would continue to fail, and if it can be corrected by restarting OctoPrint, then OctoPrint /can/ do something to correct it. Whether it /should/ do something is of course another issue.

rurwin, 2016-10-27:
Specifying the port manually does seem to avoid the failure.

foosel, 2016-11-07:
> However, does this explain why the situation is corrected by restarting OctoPrint, but not the Pi or the printer?

Hm... possibly, if you have set up OctoPrint to autoconnect to known port (+ baudrate) on server start (which would then probably also cause a reset of the controller). But other than that, no, it wouldn't explain that.

Then again, the whole "auto detect port" thing is horribly finicky and unreliable and something I seriously need to get rid of sooner rather and than later (or improve on, although that's tricky without having the auto detection take ages).

The question is, what to do with this ticket. I can't really fix this issue, since I can't do much else than what I'm already doing on a reconnect (probe with programmer mode if "AUTO" was selected, otherwise do a regular reconnect to the serial port if it was manually chosen, and the latter appears to work fine). If the controller accepts a programmer mode probe on a clean connect, but not after reporting an error, there's not much for me to do considering I have no way to differentiate between the two.

Does your printer controller reset on a serial connect?

rurwin, 2016-11-07:
> Does your printer controller reset on a serial connect?
That's a very good question and I would not like to spoil it with a bad answer.
I have no idea, and I couldn't easily find out. It's a Rambo running a bespoke Repetier, if that helps.

Seeing that you do not wish to hack the serial code further without rewriting it, (having taken a look at it I can see why,) and there is a perfectly good work-around, I think this ticket can probably be closed. Or maybe just leave it open with a very low priority.

There is one possibility. If it is easy then maybe you could do it, but do not feel you have to. If there is only one available serial port then that is the one that the printer must be on. You don't have to probe for it.

foosel, 2016-11-08:
> I have no idea, and I couldn't easily find out. It's a Rambo running a bespoke Repetier, if that helps.

A quick google search hints at it resetting. Including a complaint about that ;)

In that case I'm even more confused as to why it is showing that behaviour, I'd expect it to reset then on initial attempt to enter programmer mode during auto detect and hence not hiccup. But who knows what's going on there on the bare-metal-layer or slightly above it.

> If there is only one available serial port then that is the one that the printer must be on. You don't have to probe for it.

This still means I might have to probe through every single baudrate to figure out if it might be actually a printer (unless the baudrate is set manually), but it would indeed probably not make much difference. I'll look into that. Will probably put that into a new ticket though (I'll jot down a note though to make sure to reference this one so there's a paper trail).

I still hope to finally find the time and necessary calmness to finally switch over to the new modular communication layer approach I've now been working of and on on for the past two years. Trying to accommodate the multitude of firmware variants in the wild from one implementation is a sure way to drive oneself insane.

Anyhow... closing this but promising to create a new ticket for making the auto detection more solid in general (and ideally getting rid of that programming mode approach altogether, since it doesn't even work with a lot of controllers).

closed 2016-11-08 by foosel

rurwin, 2016-11-08:
It does seem to work with manual port selection but auto baud rate, so you have a chance. ;)