OctoPrint/OctoPrint #1555: Feature request: Display ip address on web interface

LuisDiazUgena, 2016-10-23:
Hi!

First of all, nice project :)

It will be great to be able to see the ip address on the console, to be able to ssh to the octoprint server. I've found some times that windows 10 it's failing to connect when doing ssh like:

`ssh pi@octopi.local`

Thanks for all your work!

jneilliii, 2016-10-23:
I wrote a plugin that may work for you.  You can find it [here](https://github.com/jneilliii/OctoPrint-ipOnConnect).

SpectreGadget, 2017-10-19:
This would be great in the Octoprint plugin repository!  I had to recreate my Octopi and was looking for this there.

jneilliii, 2017-10-22:
The detailed progress plugin has this functionality included, you could just use that one. I could submit to the plugin repo, but won't get to that for a little while.