OctoPrint/OctoPrint #1558: [Brainstorming] Print progress based on filament used

I-am-me, 2016-10-24:
Hi everyone,

I was thinking about a way to make print progress report more accurate. As of today most softwares uses bit based prediction, the software compares the place of the gcode line it is at to the total file size to get an approximation of the print progress. This method has flaws, the major one being, in my sense, how it is fooled by curves. Because curves are made of multiple tiny straight lines they will appear as big chunks of data (and thus a lot of progress) whilst they do not represent a significant progress in the object itself.
This is why I thought about a filament based method. The software could use the E value at the end of the gcode line and compare it to the "total filament used" line to get an approximated value of the print progress.

Exemple : 
gcode line says : G1 X86.078 Y97.369 E2664.35903
Total filament used line says : 5.565m
Progress = 2664.35903/(5.565*1000) = 0.478 = 48%

Pros : 
- More accurate than bit based guess (will not be fooled by curves)
- Relatively easy to implement

Cons : 
- Will not work on firmwares using relative or volumetric filament usage
- Needs to be able to find the line where total filament usage is written (differs between slicers, units might differ as well)

What do you think about this ? Is it a good idea ? 
Does it belong to a plugin ? 
Does it have more flaws than what I thought ?

Salandora, 2016-10-24:
Well your first and second con can easily be solved:

**Will not work on firmwares using relative or volumetric filament usage**
Basically you need to analyse the file before printing, then just sum up all the E moves, it doesn't matter it it is + or -, this way you'll have a total length/volumetric length variable.
Then inside your Plugin you need to track the E moves again and sum up all the moves again.
Shouldn't matter if it is relative or volumentric, the only thing you need to check if it is relative or volumetric but for both there exists a Mcode (Volumetric e.g is indicated by M200 D<dia>).

**Needs to be able to find the line where total filament usage is written**
Don't try to find a pattern for each and every Slicer, so as in point 1. analyse the gcode and sum up all the E values and store it, or try to rely on the already implemented gcodeInterpreter.

This is atleast a start point if I'm not total misunderstand the cons :-)

I-am-me, 2016-10-24:
I did not think about analyzing the file to sum up the lengths, it's a good idea and solves the two issues.
Thanks a lot @Salandora

foosel, 2016-10-24:
Another con: Doesn't work when printing from SD. The only information available then is current byte in file vs total.

It also can only overcome the cons mentioned already by analysing the file first (even if you are just scanning the file for specific comments you still might need to scan it fully line by line because the interesting stuff might be at the end, depending on the slicer), which means it can only work if the user is patient enough for that to take place before starting (a lot of people upload a file, immediately press on print and the complain about bad print time predictions...).

Which is why if that was tackled as a way to improve progress accuracy it would need to be done as a plugin - the core implementation needed to work always and if I'm quite honest I don't want to increase the code complexity in that place even more by  adding more "if that kind of data is available use that, otherwise do x, y or z" :)

I'm not so sure btw that measuring extrusion is the best way to go here. If your goal is to have the progress bar to reflect the actual progress over time, you'd need to take time spent on the moves into account, since even lines with little or even no extrusion can take a long time to be executed. That will cause issues though without knowing acceleration settings and without a way to get back any speed modifiers currently in place in the firmware (eg "120% feed rate" set on the LCD).

I-am-me, 2016-10-24:
You're right, it would have to scan the file in its entirety which would take a lot of time considering the enormous amount instructions in any gcode file (hundreds of thousands of lines).

I posted this as an idea to see if it was possible to get a better approximation of the print progress but from what you've told me it would be really challenging and impractical to do, and in the end it would get a result that would not be that accurate because it does not take into account the time spent on each move.

All of this to say it's not the miracle idea that will solve the print progress accuracy. At least I tried ^^

closed 2016-10-24 by I-am-me

foosel, 2016-10-24:
I'm happy for any discussion on this matter, I really didn't want to discourage you here! Sadly the situation is a quite tricky one with the available amount of information :/

I-am-me, 2016-10-24:
I'm not discouraged, the goal of this brainstorming post was to see if this idea was relevant and/or implementable. But it's not, that's a thing that happens, sometimes you have an idea that you think is good and when you speak with others and they point out the difficulties and problems (it's what brainstorming is for).
It was just an idea I had, not something I was thrilled to develop, I was just throwing the idea to see if it would be a valuable option.
Thanks for your help.