OctoPrint/OctoPrint #1559: [Feature Request] Resuming print executes stack of movements done since pause

nate-ubiquisoft, 2016-10-24:
#### What were you doing?

Pause a print (i.e. to change filament) then use the **control** screen to move the hotend away from the print while changing filament, then press **resume**. Hotend moved back to the correct X-Y orientation but the Z change was not reverted.

#### What did you expect to happen?
Resuming print would move hotend back to where it was when the **pause** executed.

#### What happened instead?
Printer begin printing 30mm above the printed object's pause point

#### Branch & Commit or Version of OctoPrint
1.2.16

#### Printer model & used firmware incl. version
Lulzbot TAZ 6 with most recent firmware

#### Browser and Version of Browser, Operating System running Browser
Chrome, Firefox, Safari (not browser specific)

Omitting remaining details as they do not pertain to [Feature Request]s.

I have read the FAQ. [YES]

foosel, 2016-11-26:
Will be possible in 1.3.0 (and can already be tested in 1.3.0rc2). Due to some firmware limitations though it requires manual configuration of the Pause and Resume GCODE script though. An example is provided in the [1.3.0rc2 release notes](https://github.com/foosel/OctoPrint/releases/tag/1.3.0rc2) (also [here](https://gist.github.com/foosel/1c09e269b1c0bb7a471c20eef50c8d3e)).

nate-ubiquisoft, 2016-11-26:
Awesome thanks for the detailed response! Love OctoPrint and will keep doing everything I can to get everyone running it! 
Also, unrelated, I hope to publish my plugin soon (some components are in a java war project and need to be migrated to python) that will allow prints to be scheduled by the estimated end time and start automatically to end at the specified time.

closed 2016-11-26 by nate-ubiquisoft