OctoPrint/OctoPrint #1560: Server initialization fails silently when logging.yaml is empty

brianholley, 2016-10-25:
#### What were you doing?

Restarting the octoprint server with an empty logging.yaml (after removing event debug level). The service fails to start and nothing is written to octoprint.log.

Minimal commands to reproduce:
mv ~/.octoprint/logging.yaml ~/.octoprint logging.yaml.bak
touch ~/.octoprint/logging.yaml
sudo service octoprint restart

#### What did you expect to happen?

Service to restart correctly or log the appropriate error.

#### What happened instead?

The service fails to start and nothing is written to octoprint.log.

#### Branch & Commit or Version of OctoPrint

1.2.16 (master branch)

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

octoprint.log is empty
Console output when running octoprint process manually: https://gist.github.com/brianholley/e3d891d9278a2bc294baae9a14d95085

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

brianholley, 2016-10-25:
I'm happy to submit a PR to fix this - my quick testing shows that just catching the exception coming from logging.config.dictConfig in _setup_logging is enough to allow the server to start as usual. Unfortunately, the logger isn't started at that point, so I can't write any information about the failure to parse logging.yaml to octoprint.log.

foosel, 2016-10-25:
@brianholley happy for any pr against devel, will take care to backport to maintenance. Obviously I never tested with an empty logging.yaml, only with a filled one and without one m)

(Short heads up, merging might take a couple of days, I'm currently not in working order)

foosel, 2017-02-22:
Fixed on `maintenance` (where it actually was already fixed in a way, made it more resilient though), will merge to `devel` ASAP, will also be part of 1.3.2

commit 79af90dd (2017-02-22): Make sure we don't try to use an empty logging.yaml

Fixes #1560
closed 2017-03-16 by foosel