OctoPrint/OctoPrint #1563: GCode to Pause For Color Change At Layer Delayed

jpvonhemel, 2016-10-26:
#### What were you doing?

Printing a design with the following GCode inserted just after the Layer 17 Comments.  The purpose is to change filament colors at this point.  

Here is my gcode file..

<script src="https://gist.github.com/jpvonhemel/d29c70a13eeac746a10ad2ac09679879.js"></script>

Here is a snippet of the code I use to change filament.   I changed M25 to M0 to pause in Octoprint.

`; layer 17, Z = 3.38
; inner perimeter
; outer perimeter
G91 ; set relative positioning
G0 Z40 ; raise Z (drop bed)
G90 ; set absolute positioning
M84 S0 ; disable idle timeout
M0 ; Pause SD Print
G92 E0 ; zero extruder
G1 X111.049 Y193.306 F18000
G1 Z3.380 F1200
G1 E0.0000 F2400
G92 E0
G1 X111.049 Y186.604 E0.2006 F2880
G1 X111.316 Y186.604 E0.2086
G1 X111.316 Y189.193 E0.2862
G1 X111.667 Y189.142 E0.2967
G1 X113.292 Y189.142 E0.3454
G1 X114.625 Y186.604 E0.4312
G1 X115.041 Y186.604 E0.4437`

#### What did you expect to happen?

Pause as this code is received at the beginning of Layer 17

#### What happened instead?

The pause occurred about one quarter into the 17th layer.   It seemed like it was delayed from octoprint or the printer receiving, not sure which.      The code is at 40585 but octoprint sent commands well past that before the printing paused.

#### Branch & Commit or Version of OctoPrint
Version: 1.2.16 (master branch)

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version
Makergear M2 Dual Firmware

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 53.0.2785.116 (64-bit)

[If applicable, always include if unsure.]

#### Link to octoprint.log

<script src="https://gist.github.com/jpvonhemel/98f71aab1dc6bc7b4fd5407aad1f4f89.js"></script>

#### Link to contents of terminal tab or serial.log

Not sure where this is

#### Link to contents of Javascript console in the browser

It is blank

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

jpvonhemel, 2016-10-26:
Here is my gcode file. 

[https://www.dropbox.com/s/9r0dpyiu8rpolrc/Abvisor_Furhman.gcode?dl=0](url)

foosel, 2016-10-26:
> **Link to contents of terminal tab or serial.log**
>
> Not sure where this is

There:

[image]

And since it is a ticket related to printer communication, this information is absolutely mandatory. See also this segment from [the contribution guide](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#where-can-i-find-those-log-files-you-keep-talking-about):

> It might happen that you are asked to provide a more thorough log of the communication with the 
> printer if you haven't already done so, the serial.log. This is not written by default due to performance 
> reasons, but you can enable it in the settings dialog. After enabling that log, please reproduce the 
> problem again (connect to the printer, do whatever triggers it), then copy-paste 
> ~/.octoprint/logs/serial.log (Windows: %APPDATA%\OctoPrint\logs\serial.log, MacOS: 
> ~/Library/Application Support/OctoPrint/logs/serial.log) to pastebin or gist and include the link in 
> the bug report.

I need the communication log from a couple of lines before the M0 in the file to a couple of lines after it has triggered. You might be able to retrieve that from the terminal tab if you keep an eye on things, otherwise you'll need to enable `serial.log` via Settings > Serial, run your print with that (warning, performance and hence quality impact) and then provide the resulting log file.

Also, please always provide `octoprint.log`. Never file a bug report without it, it contains important information for debugging (such as your installed plugins, state changes etc) and is essential for being able to look into your problem in the first place.

jpvonhemel, 2016-10-26:
Sorry new to this process.   This occurred last night, so I think I lost the data.  I print this quite a bit, so I will try to reproduce and give you the data you need.   

I checked here:  Log communication to serial.log (might negatively impact performance).   Will this file be available under logs?

jpvonhemel, 2016-10-26:
The octoprint.log link did not paste in for some reason.   Here it is.

https://gist.github.com/jpvonhemel/98f71aab1dc6bc7b4fd5407aad1f4f89

foosel, 2016-10-26:
> I checked here: Log communication to serial.log (might negatively impact performance). Will this file be available under logs?

That's correct and yes, it will. That warning should be taken seriously though, so while that is enabled don't print stuff that is dear to your heart, it might turn out a bit blobby ;)

It would actually even be best if you could try if you can reproduce the issue with a minimum example (e.g. a small cube sliced with a pause right after the first layer). The smaller the example with which you can repro, the better for debugging purposes.

> [octoprint.log]

Thanks! In that case it doesn't contain any hints. Was worth a shot though. Waiting for `serial.log` it is :)

jpvonhemel, 2016-10-26:
No, thank you very much.  I greatly appreciate your work!

jpvonhemel, 2016-10-26:
Printing a small sample and still expect the M0 to stop the printing at Layer 17.   I am noticing the GCode Viewer is jumping around layers, but suspect it is due to logging the serial connection.

jpvonhemel, 2016-10-26:
The sample stopped correctly twice.    I wonder if the GCode Viewer showed it was on Line 17, when it really wasn't.     I will continue to monitor this.

jpvonhemel, 2016-10-26:
Here is the link to octoprint.log

https://gist.github.com/jpvonhemel/b4f7ca4ec291e4c3aa6715edaa5a5d4c

Here is the link to serial.log

https://gist.github.com/jpvonhemel/3298cff12c1c965288720a8e5f4f12b6

Is URL the best way to add the files?

ntoff, 2016-10-27:
line 40589: M0 ; Pause SD Print

M0 isn't used to pause SD printing.

http://reprap.org/wiki/G-code#M0:_Stop_or_Unconditional_stop

>The RepRap machine finishes any moves left in its buffer, then shuts down.

That's M25 http://reprap.org/wiki/G-code#M25:_Pause_SD_print

foosel, 2016-10-27:
> Is URL the best way to add the files?

You can also attach txt files to tickets (see the "Attach files" line right below the editor window when commenting).

> Here is the link to serial.log

So from the looks of that it paused exactly at where your `M0` was placed:

```
2016-10-26 20:00:37,479 - SERIAL - DEBUG - Send: N9678 G1 X111.230 Y131.892 E0.6644*98
2016-10-26 20:00:37,571 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,573 - SERIAL - DEBUG - Send: N9679 G92 E0*118
2016-10-26 20:00:37,579 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,581 - SERIAL - DEBUG - Send: N9680 G1 E-1.0000 F2400*40
2016-10-26 20:00:37,613 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,615 - SERIAL - DEBUG - Send: N9681 G91*39
2016-10-26 20:00:37,620 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,623 - SERIAL - DEBUG - Send: N9682 G0 Z40*98
2016-10-26 20:00:37,711 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,713 - SERIAL - DEBUG - Send: N9683 G90*36
2016-10-26 20:00:37,719 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,721 - SERIAL - DEBUG - Send: N9684 M84 S0*111
2016-10-26 20:00:37,726 - SERIAL - DEBUG - Recv: ok
2016-10-26 20:00:37,728 - SERIAL - DEBUG - Changing monitoring state from 'Printing' to 'Paused'
2016-10-26 20:00:37,734 - SERIAL - DEBUG - Send: M105
2016-10-26 20:00:37,747 - SERIAL - DEBUG - Recv: ok T:220.0 /220.0 B:69.3 /70.0 T0:220.0 /220.0 T1:29.5 /0.0 @:56 B@:127
2016-10-26 20:00:42,737 - SERIAL - DEBUG - Send: M105
2016-10-26 20:00:42,750 - SERIAL - DEBUG - Recv: ok T:220.2 /220.0 B:69.0 /70.0 T0:220.2 /220.0 T1:29.5 /0.0 @:51 B@:127
2016-10-26 20:00:47,738 - SERIAL - DEBUG - Send: M105
2016-10-26 20:00:47,751 - SERIAL - DEBUG - Recv: ok T:220.7 /220.0 B:68.9 /70.0 T0:220.7 /220.0 T1:29.5 /0.0 @:40 B@:127
2016-10-26 20:00:52,736 - SERIAL - DEBUG - Send: M105
2016-10-26 20:00:52,749 - SERIAL - DEBUG - Recv: ok T:220.7 /220.0 B:69.0 /70.0 T0:220.7 /220.0 T1:29.5 /0.0 @:40 B@:127
2016-10-26 20:00:57,737 - SERIAL - DEBUG - Send: M105
```

Note the "Changing monitoring state from 'Printing' to 'Paused'" right after the `M84 S0`, which is the last line before your `M0`. Note that the `M0` itself is NOT sent since it is only commonly used for inline pause commands and if sent to the printer would actually bring it to something akin to an emergency stop - therefore OctoPrint interprets it as "please pause here" but doesn't send it over the line.

But from the looks of that log, all is well.

>  I wonder if the GCode Viewer showed it was on Line 17,

Wait, do I understand you correctly that your statement that "The pause occurred about one quarter into the 17th layer" up there was refering to what was displayed in the GCODE viewer, NOT what the printer in fact was doing? Because that would be a completely different (and way less critical) issue from what I so far understood this to be.

jpvonhemel, 2016-10-27:
forgot to change the comment.

jpvonhemel, 2016-10-27:
Thank you for looking at my log files.   I think you can close this ticket.    If I notice issues with the GCode viewer again, I will take some screenshots and open a new ticket.   

One question first, the line number in the gcode file should match the line number displayed in the Gcode views, correct?

foosel, 2016-11-07:
> the line number in the gcode file should match the line number displayed in the Gcode views, correct?

**line** number or **layer** number? Please do not confuse these. They are completely different things.

If **line** number: I have no idea what you are talking about, the GCODE viewer doesn't display line number sent to the printer. In any case the line numbers from your file do not map in any way to what is actually sent to the printer, since OctoPrint will strip empty lines and comments from the file before sending it (the serial line is slow, we need so save bandwidth there), and will also send commands on its own to track the printer's vitals (e.g. an `M105` every couple of seconds to plot temperatures).

If **layer** number: "Layer 17" will not necessarily indicate a z-value of 17 (unless you are printing with 1mm layer height). The GCODE viewer will go through the GCODE file, look for moves with extrusion and the Z values of those sorted from smallest to biggest are the layers (and the layer number) it will display. "Layer 17" will be whatever is in your GCODE file with the 17th-lowest Z value.

If your question is referring to any comments placed in your GCODE file by your slicer, no, that does not necessarily need to match up with what the GCODE viewer displays. Every slicer has its own comment format and OctoPrint doesn't try to parse any of that in any way to determine layers or layer numbers - no standard, no way to do that reliably.

Also closing this as suggested by you. Note that you can still comment.

closed 2016-11-07 by foosel