OctoPrint/OctoPrint #1564: Read/Write filesystem required after 1.3.0 rc1 installation

sanchosk, 2016-10-27:
#### What were you doing?
I installed the 1.3.0 rc1 release onto my system.
The system is:
- Orange Pi One
- Running from SD card in read-only mode
- OctoPrint work folder (with logs, gcode, timelapses, ...) is mounted using NFS in R/W mode

During installation, I remount the root filesystem to R/W mode. I do the installation and right after, when recommended by the frontend, I not only restart the OctoPrint, I reboot the whole system.
This makes the filesystem read only again.

#### What did you expect to happen?
I expected the same thing as before - the new version starts and works no problem after the reboot.

#### What happened instead?
After reboot the OctoPrint did not even start - it wanted to do some PIP installations to finish the update. As a result (as pip was unable to write to disk) the OctoPrint was not able to run at all.
I had to re-mount the system to R/W, run OctoPrint, let it install the remaining parts and then do another reboot to read-only mode.

#### Branch & Commit or Version of OctoPrint
1.3.0 rc1 (rc/devel branch)

#### Printer model & used firmware incl. version
RepRap serial connection, /dev/ttyUSB0, Repetier firmware 0.98

#### Browser and Version of Browser, Operating System running Browser
Firefox, not sure what version did I used then, I upgraded system in the meantime :(

#### Link to octoprint.log
I have to provide this later as I don't have remote access right now.
[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log
See above, will provide that later, although this one is probably not necessary.

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:
N/A

sanchosk, 2016-10-27:
Here is the log:
https://gist.github.com/sanchosk/3e8c369d3ad9c76f5572bab0f4be5a09
Unfortunately, as you can see, the OctoPrint did not even wrote anything to log - there is a huge gap between the update happening at 12:36 and until I fixed the read-only filesystem, nothing got even to the log. The next log starts at 12:49, which was my first start with RW system. Although I've seen a cli output about some download of plugins or something (cannot recall that exactly), it's not in log.
I can try to do a repeatability test - build it on my own system from 1.2.17-rc4 and upgrade to 1.3.0-rc1 if that helps...

sanchosk, 2016-10-27:
Strange. I've build OctoPrint 1.2.16-rc4 (from maintenance) on my workstation.
I re-mounted the virtual fs in read-only mode and run it - all OK.
I re-mounted to R/W and upgraded to 1.3.0-rc1 (devel). After successful update, I quit it the same way as before, remounted to R/O and started.
And the error reported above did not happen :(
I will have to dig my "trash" box and get an old banana pi and replicate what I had on Orange Pi One here.
Is it possible the error was caused not by OctoPrint, but by some plugin installed? I have print history, repetier eeprom and printer stats...

foosel, 2016-10-27:
I just de-armed the bot - your ticket template missed one crucial line which made it comment, should be fine now I hope.

About the issue, OctoPrint should not attempt to pull in packets after an update, all dependencies should actually be installed during an update right away (the whole update script should be self contained so to say). So I'm a bit at a loss here what could have happened.

It shouldn't be a plugin, although I of course can't guarantee that since I don't control the plugins, but for the three you are listing it would strike me as odd if they did something like this.

You are saying "it wanted to do some pip installations to finish" on initial restart - how did you come to that conclusion? Some error message on startup? If so what? And what dependencies? That might be the crucial hint here...

foosel, 2016-11-25:
@sanchosk still waiting on some more info here :)

sanchosk, 2016-11-25:
I tested it twice more, none of it showed any behavior like the one I've experienced during the update :( All of the tests ended with octo starting directly. Perhaps this was some sort of network failure. I'm closing it. I tried OrangePi One, Banana Pi and virtualBox VM on my notebook - all OK :(
Can't replicate => closing.

closed 2016-11-25 by sanchosk

foosel, 2016-11-25:
I'd rather see that as a ":)" instead of a ":(" though ;) Thanks for the update!