OctoPrint/OctoPrint #1569: [Request] Temperature upper limit / cutoff for input fields - ingore temps outside range.

Cyberman-tM, 2016-10-31:
This is of course to prevent damage due to user error. I just caught the bed getting (too) hot because I probably entered 200 degrees instead of 40 (the nozzle is set to 200).

My fault, and no one to blame but me. Still, it would be useful if the program could "double check" my input (not the gcode, that'd go too far).
I'll probably never set the bed to anything beyond 100 (even this is rare), so there is no reason for it to ever go beyond 100°.

Would be nice to have, IMO.

nophead, 2016-10-31:
It would have to be configurable as I use temperatures as high as 148C. My
firmware sets a limit of 150C.

derpicknicker1, 2016-11-01:
I think this is a MUST HAVE feature of the FIRMWARE and nothing octoprint has to care about.

nophead, 2016-11-01:
Yes all the physical limits of the machine should be set correctly in the firmware, not the slicer or the host.