OctoPrint/OctoPrint #1570: Pause for ~30 seconds during USB print

brycedjohnson, 2016-11-01:
#### What were you doing?

Printing a small model

#### What did you expect to happen?

No pauses, print to finish.  

#### What happened instead?

Had around a 30 second pause during the USB print.  In the terminal it was stuck on this line:

Send: N1131 G1 X-11.581 Y-25.722 E1.9993*103
Recv: ok
**Send: N1132 M105*22**

I was NOT actively using octoprint.  I've had a few hesitations when I'm clicking around octoprint during a print, but these long ones seem like something else....  I've had several that I noticed and large blobs of plastic where it has happened when I didn't notice.

#### Branch & Commit or Version of OctoPrint

 1.2.16

#### Printer model & used firmware incl. version

Kossel Pro (delta printer) running marlin.  Using a decent looking USB cable with a ferrite bead

#### Browser and Version of Browser, Operating System running Browser

Windows 10 with Chrome 54.0.2840.71 m (64-bit)

#### Link to octoprint.log

https://gist.github.com/brycedjohnson/0fdeb9e5d4ac6c19bc43b2b26ed9c15b

#### Link to contents of terminal tab or serial.log

Just copied these lines: I can get a longer capture next time...

Send: N1131 G1 X-11.581 Y-25.722 E1.9993*103
Recv: ok
**Send: N1132 M105*22**

I just noticed the fake ACK in advanced options.  Would that be good to try?

foosel, 2016-11-02:
OctoPrint won't send any lines to your printer unless it receives an ok back from it telling it that it's OK to do so. 

From what I understand from your report your printer took its sweet time replying to the M105 line there. The question is why your printer pauses in replying here. OctoPrint is behaving correctly in waiting for the reply (if it didn't it would risk overflows in the printer firmware that would cause issues ranging from skipped lines to full out controller crashes). Pressing the "fake ack" button might help (or not, of the printer is fully unresponsive during that down time), but will only mask the issue.

You really need to figure out why the response is not coming or coming with such a huge delay. Printer controller getting hot? What controller are you running? Is it reproducible in certain files?

brycedjohnson, 2016-11-04:
Thanks the the info, that makes sense.  I am using a brainwave pro on kossel pro.  I've had it happen shortly after starting a print so it shouldn't be too hot yet... Maybe there is a communication issue at time.  I'll dig in an see if I can see what is happening when printing from USB.

closed 2016-11-04 by brycedjohnson