OctoPrint/OctoPrint #1571: Forwarding IPv6 connection receiving a cookie error

fang64, 2016-11-02:
#### What were you doing?

I bound OctoPrint to IPv6 via HAProxy and received a:
CookieError: Illegal key value: session_Pa000:1210:8121::3]

#### What did you expect to happen?
The OctoPrint page should have loaded

#### What happened instead?
Internal Error 500 returned by the OctoPrint server. 

#### Branch & Commit or Version of OctoPrint
Version: 1.2.16 (master branch) 
354042b84d488db38ac1917bb69bd164f4e7f750

#### Printer model & used firmware incl. version
Solidoodle 2 Pro, Marlin Firmware

#### Browser and Version of Browser, Operating System running Browser
Firefox 49.0.2

#### Link to octoprint.log
https://gist.github.com/fang64/bef5acb77f99654911b28c87d341c0c7

foosel, 2016-11-02:
Can you please provide the haproxy configuration that you used for that? It looks like something is wrongly sent in the forwarding headers, causing part of the address to be interpreted as a port number and causing that issue.

Also, please never leave out parts from the ticket template.

fang64, 2016-11-02:
@foosel, I apologize for that, here is the gist of that.

https://gist.github.com/fang64/cabddca7e8bbe56fc86c26eed44ac3dd

foosel, 2016-11-07:
Fixed in `maintenance` and `devel` branches. Will NOT make it into 1.2.17 but will be included in 1.2.18 and also the next RC for 1.3.0.

commit 2952b262 (2016-11-07): Do not hiccup on ipv6 addresses in the HTTP Host header

Fixed port extraction. Now supports hostname, hostname + port,
ipv4, ipv4 + port, ipv6, ipv6 + port

Fixes #1571
closed 2016-11-30 by foosel