OctoPrint/OctoPrint #1572: [request] Prioritise login before other resources

on-d-mand, 2016-11-02:
Just a minor suggestion/request would be to let the user auto login code run earlier in the process... When I notice a print is failing I open the browser to octoprint and since (in my case) the Pi is rather busy it can take up to a minute or more before I can even cancel the print or make proper quick temperature adjustments.... When there is no print running it takes approximately 35 seconds

Looking at the XHR through Chrome I see:

sockjs/info (?)
api/files (the files)
api/printerprofiles (the printer profiles obviously)
api/slicing (the configured slicer)
api/timelapse (timeplase plugin)
api/connection (printer connection)
api/printer/command/custom (custom printer controls)
ap/settings (the settings)
plugin/printhistory/history (the print history)
api/slicing/cura/profiles (cura slicing profiles)
api/login (<<<<<   finally the login)
api/users (the available users)
api/logs (the log files)
plugin/announcements/channels (news feeds?)
api/plugin/pluginmanager (?)
plugin/softwareupdate/check ( check for updates)

Some of these resources seem really unrequired for the user to login... preferably it would be something like

api/connection (printer connection)
api/printer/command/custom (custom printer controls)
ap/settings (the settings)
api/login (<<<<<   finally the login)

and then the rest can be loaded thus enabling someone to intervene faster on the printer
----

#### What were you doing?

Print a file

#### What did you expect to happen?

Login quickly and stop the print

#### What happened instead?

Waited more then a minute until I could cancel it

#### Branch & Commit or Version of OctoPrint

Version: 1.2.16 (master branch)

#### Printer model & used firmware incl. version

Zonestar Prusa i3 (clone)

#### Browser and Version of Browser, Operating System running Browser

Google Chrome (Version 54.0.2840.71 (64-bit)), OSX Siera 10.12.1

#### Link to octoprint.log

(not really required?)

#### Link to contents of terminal tab or serial.log

(not really required?)

#### Link to contents of Javascript console in the browser

(not really required?)

#### Screenshot(s) showing the problem:

(not really required?)

I have read the FAQ.

Yes

shusain, 2016-11-04:
@on-d-mand which version of the pi are you using? Just wondering because here on a Pi 3 (not overclocked, on wifi) when it's idle the page DOMContentLoad event happens in around 900ms here and all the XHR requests are done within 3-4seconds, 30 sounds really excessive.

Have you checked out what's going on processing/memory wise on the pi?  If you haven't already try sshing to the pi and checking top/htop for cpu usage and "free -h" for memory usage to see which is the bottle neck or if it's some other process outside the web server or octopi stuff that is slowing the system down.

on-d-mand, 2016-11-04:
We experience this on a Pi 1 and Pi 2, clean install from the Octopi image and only some additional plugins.. It is on a rather cheap Wifi card, maybe that is the issue but I think loads of Pi users could benefit from a better order and seems like a rather small fix.

foosel, 2016-11-04:
>  cheap Wifi card, maybe that is the issue

Very probable actually

> seems like a rather small fix

It isn't I fear

shusain, 2016-11-04:
Yeah regarding it being a "small fix" I do web dev as well and especially with something like this where there is a lot of XHR going on already for things and socket connections etc. it becomes pretty difficult to really stagger out the loading of things properly and make sure that whatever you need loaded is loaded by the time you start using it without just loading a lot of things up front (it's possible but not an easy effort to lazy load things and make sure everything still works)  Much easier to replace a Pi :)  I was looking at maybe reducing the communication since it seems there is a lot of traffic in the web socket connection and if I am logging in remotely to check on the video feed all that extra data really adds up (on a not unlimited cell connection), but again not trivial to do without losing functionality I'm sure and spending some time digging into the code.

nophead, 2016-11-04:
When I first started using OctoPrint on an RPI B with only a class 4 SD card I don't remember start up time being an issue. It seems to have got a lot slower over time.

shusain, 2016-11-04:
@nophead by startup time do you mean the boot time or the time to access the page in a browser? If you're talking in the browser could do a timeline and/or profiling comparison of an old build vs a new build I suppose and see what the major differences are.  On boot time there's an app think called bootchart that will outline the startup times of various services.

foosel, 2016-11-04:
@nophead When you first started using OctoPrint the UI was also way less complicated to render on the backend, and there were far less checks in place to compensate for user error overall ;)

shusain, 2016-11-04:
@nophead This is what it looks like here:
[image]

I wonder if that time seems high to you all as I have only started using OctoPi recently.  It really seems acceptable to me to wait about 4 seconds on initial load for what it does, but always in for performance tuning :)

To see the same view on your computer in Chrome hit F12 then go to timeline tab and check the boxes for network etc. and refresh the page.

If you/we know what requests are taking the longest or taking much longer than they used to could either diff things or use valgrind or whatever to get at what's going on under the hood.

OctoPi Version 1.2.16 (master branch)
# uname -a
Linux octopi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

It's a RasPi 3 Model B version 2

on-d-mand, 2016-11-04:
Startup time is not the main concern, that's fine... but login (and thus enable the controls) happens for you after 4 seconds and that's fine but as you can see it can be within 1 or 1.4 seconds if the order of the XHR calls is done differently. You can clearly see it hit the max open connections on the webserver (6 I guess from your screenshot) and it only gets freed after those have done... Also you can see Timelapse beeing more important to in the code then the login. Also try this during print which could make a big change.

Of course I could buy a new Pi3 which is blazingly fast but beyond this issue there is no speed issue with octoprint and the print itself thus seems like a more overkill.

nophead, 2016-11-04:
A forced refresh takes around 30s and a restart of the RPI takes minutes.

foosel, 2016-11-04:
> A forced refresh takes around 30s

Yes. Because then you tell both OctoPrint and your browser to throw everything away they both might have in their caches and make OctoPrint generate everything anew, and that takes a bit because complexity. The server side cache and caching headers for client side browser caching are in place to prevent this from happening on every reload, because stuff can take long (and the preemptive cache on server startup is in place to make sure the page is rendered ideally *before* the user tries to access it for the first time for this very reason). A regular reload will allow utilisation of the cache and you will see timings like the above (or even better in 1.3.0rc1). 

Complaining about a force refresh taking long is like complaining that your car is slow when you keep the brakes engaged ;)

And about the boot up times, first make sure it's actually the server startup you are measuring and not the underlying OS (which I can't do anything about), and then keep in mind that OctoPrint also now has to initialise a plugin system and do the preemptive caching mentioned above, plus a bunch of cleanup tasks for better resilience. I think we talked about this elsewhere but: it is a server built for long uptimes. I'd rather have it start up a bit slower and then save the user time once it is up and running than starting semi initialised but blazingly fast and then lazy loading everything once a user tries to use it, making them wait then.

But we are not actually talking about initial startup or force refresh loading times here but the general order of requests from the client and how that might be optimised. So let's stop the derail here ;)

The issue with that order optimisation is that yes, it might be possible to get better reaction times here, but how things are currently triggered (read: automatically through a mini framework, agnostic of the individual components, coupled through named callbacks), I fear it is not a trivial task to influence the loading order after the initial settings call in order to prioritise something like eg the login request.

So while I'll happily look into this, it will have to wait until after 1.3.0 stable, and if its a pressing issue for you @on-d-mand, I'd rather look into the wifi throughput, make sure those dongles are producing proper data rates. The loading times currently observed sound too slow than what I'm experiencing on my own (similar) setup, and this might be a low hanging fruit as a pain relief.

foosel, 2017-06-29:
I've just pushed a change to `maintenance` that changes initial request order in such a way that not only the passive login is now prioritized but hopefully the enforced ordering of that before everything else should also fix #1881 (if that was caused by what I think it was, haven't been able to reproduce).

Already on `maintenance`, soon `devel`, will be released with 1.3.5. Testing (either now against the branches or as soon as a first release candidate is out) is strongly welcome - this was a quite fiddly change and the more people who take this for a ride before it hits everyone, the better.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel