OctoPrint/OctoPrint #1573: [1.3.0rc1] WebSocket Error

FinstererFred, 2016-11-02:
#### What were you doing?
Updated to 1.3.0rc1 and  tried to connect to octoprint via internet.
The ocotoprints port 80 is directly forwarded from an fritzbox, 
there is no proxy involved, i acccess the page via http.

#### What did you expect to happen?
To load the octoprint webpage and connect to the server.

#### What happened instead?
The page loads, but displays the message "Server is offline". 
There are several error messages in the javascript console
"Error during WebSocket handshake: 'Sec-WebSocket-Accept' header is missing"
also see Pastebin below.

Note: connecting from the local wlan works fine.

#### Branch & Commit or Version of OctoPrint
1.3.0rc1

#### Printer model & used firmware incl. version
Anet A8, Marlin v1 (Skynet3D 1.1 branch)

#### Browser and Version of Browser, Operating System running Browser
Chrome 54.0.2840.87 m, Firefox 49.0.2, Windows 7

#### Link to octoprint.log
No log files are available since the update to 1.3.0rc1

#### Link to contents of terminal tab or serial.log
Not applicable

#### Link to contents of Javascript console in the browser
http://pastebin.com/Rd3Y5Yys

foosel, 2016-11-02:
Please do not truncate the ticket template or the bot will trigger as you just saw ;)

You say you've forwarded the port in your router. So I assume you are accessing your OctoPrint instance via `http://<your external address>:80/`, is that correct? Is this on a regular unmodified OctoPi image? If so, could you please also provide the file `/etc/haproxy/haproxy.cfg` from it? If not, please further describe your setup. Finally, is there a way to take a look at this myself (you don't have to share the address publicly, just drop me an email - being able to look at what the client is sending and receiving might help pinpointing this).

FinstererFred, 2016-11-02:
Hi, you are right, i access the OctoPrint directly via http://xxx.de/
Its an unmodified OctoPi image, and i pasted the haproxy.cfg here: 
http://pastebin.com/DUhfvjXV

I will send you an email with the address shortly, thanks :-)

foosel, 2016-11-02:
So I just tried to repro this with the data you provided in the mail (thanks!) in a Chrome 54.0.2840.87 m and a FF 33 & (after an overdue update) 46.0.1, all under Win10. I cannot reproduce the problem. Frontend runs flawlessly for me, including updates via the websocket, no error messages present. To rule out that it might be caused by being logged in on a fresh connect, I also logged in with the credentials you provided and did a hard reload (and then realized that that might have been bad considering you were printing something, I hope I didn't cause any slowdowns with the reload, sorry if I did). 

In any case, whatever is acting up there is not a general issue with OctoPrint or your forwarding setup but appears to be client specific. Not being able to see it for myself of course makes it a bit tricky to take a guess, but I suggest to give it a try with a fresh browser profile to rule out any issues with your  installed extensions. And maybe also try a different machine (laptop or something?) to get an additional data point.

Btw, log files are of course still available, I was just also taking a look into your `octoprint.log` - but other than a lot of disconnects and reconnects from the same client IP there's nothing in there that might relate to the issue.

foosel, 2016-11-02:
Small addition: your print job just finished and I also did a quick test against Chrome and FF on Win7 with Browserstack. Also couldn't reproduce there.

FinstererFred, 2016-11-02:
Thank you very much for your time!

I tested the access from two other machines, and it seems that the problem only occours at my workplace (tested on 2 different computers).

So think the reason this happens is the firewall in our company and the occurence of the error and my update to 1.3.0rc1 seems to be a coincidence.

I will speak to our admin, and see if i can figure something out.

Thanks again, 
and keep up the great work!

closed 2016-11-02 by FinstererFred

foosel, 2016-11-02:
In that case I could imagine a transparent company proxy to be at fault here. You could also just do an ssh tunnel to circumvent this (be sure you don't go against company policy though!)