OctoPrint/OctoPrint #1574: OctoPrint stops printing mid print.

Brianp82, 2016-11-03:
Hey guys. So I am using Octoprint with a Raspi 3. I have had many good prints in the 3 days of using it but I also have had it stop printing randomly twice in this time. I checked my log file and I cant see anything out of the ordinary except these command right around when I think its stopping.  There are other log entries in between these but nothing that would indicate why it stopped. Any ideas? I also didnt check my Terminal tab to see if there were any gcode errors or whatever. I didnt realize I could possibly check there.

2016-11-03 03:50:16,670 - octoprint.server.api - INFO - Performing command: sudo service octoprint restart
2016-11-03 03:50:16,817 - octoprint.server - INFO - Shutting down...
2016-11-03 03:50:17,087 - octoprint.server - INFO - Goodbye!
2016-11-03 03:50:19,792 - octoprint.server - INFO - Starting OctoPrint 1.2.16 (master branch)
2016-11-03 03:50:57,441 - octoprint.server.api - INFO - Performing command: sudo shutdown -r now
2016-11-03 03:50:57,558 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.3
2016-11-03 03:50:57,615 - octoprint.server - INFO - Shutting down...
2016-11-03 03:50:57,679 - octoprint.server - INFO - Goodbye!
2016-11-03 03:51:06,091 - octoprint.server - INFO - Starting

foosel, 2016-11-03:
From that log it looks like someone (you? Someone else with access to the OctoPrint instance?) shut down your OctoPrint instance by clicking on System > Shutdown and confirming that action after restarting OctoPrint, also via the system menu and confirming that action.

It's not clear though if that actually happened while you were printing. It will not be possible to analyse this further without you providing more information, including a *full* log file and terminal excerpt. 

Basically all that information the ticket template asked for which you deleted.

Also please do not create further duplicates of this ticket either here or in the OctoPi bug tracker. Simply provide the requested information instead.

Brianp82, 2016-11-03:
Thanks  yeh I definitely didn't do the restart  my browser was closed and nobody is on my network for sure

Brianp82, 2016-11-03:
And sorry it's my first time using githu for a issue  I'll have to look for the bug template  this is all the info I have though  I didn't check the terminal tab unfortunately .

foosel, 2016-11-03:
> I'll have to look for the bug template

The bot linked you to the contribution guidelines, those include the ticket template and more information relevant for getting help.

> this is all the info I have though

Then we'll have to wait if it happens again in which case you should provide a fully filled out template. 

Based on what you provided (if the time stamp really matches your ongoing print, 3am strikes me as a bit of an odd time to be printing), this is not a bug but someone manually shutting down stuff and ignoring all associated warnings. If that wasn't you I'd strongly suggest to check your network's security, change your OctoPrint account password and if there are any port forwards from your router to your LAN to shut them down.

Brianp82, 2016-11-03:
Hey, I actually print overnight almost every night. But just so you know the log time stamp differs from the actual time. Its off by a few hours. I will change the password to my octoprint though. I have no port forward setup. So it would have to be someone on my network which there is nobody else on it for sure.

Brianp82, 2016-11-04:
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves

That is the error that is coming up in my terminal when it freezes

Brianp82, 2016-11-04:
This is in the middle of printing now but didnt stop completely. Any ideas?

Send: N9 G92 E0*78
Recv: ok 9
Send: N10 G1 E-3.0000 F3000*41
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N11 M105*23
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00
Recv: ok 10

Brianp82, 2016-11-04:
And btw I am powering the pi with a 2.5a wall adapter. I got it all from here:

https://www.amazon.com/gp/product/B01C6EQNNK/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

foosel, 2016-11-04:
> provide a fully filled out template

ntoff, 2016-11-06:
>Thanks yeh I definitely didn't do the restart *my browser was closed* and nobody is on my network for sure

Does the IP at the end of this line match your PC or the raspberry pi? Does it match any computer you have control over?

>2016-11-03 03:50:57,558 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.3

What kind of plugins do you have installed? Any that maybe restart the server at a certain time? Auto shutdown plugin which will shut down when a job is done? Anything like that?

Brianp82, 2016-11-06:
Yes that matches something I have control over. That is me logging on and off of octoprint to check the print status. I do not have any shutoff plugins installed.

ntoff, 2016-11-06:
>I will change the password to my octoprint though

Also change your API key and your password for Raspbian (assuming it's installed on a raspberry pi). You don't even need the password to shut down if you have the API key and if someone has the password for your raspberry pi (most people probably leave that as default) they have control over that entire machine.

Brianp82, 2016-11-06:
I changed the password to my Octoprint. I didnt install raspian on my pi when I got it. Just Octopi and I made a new username and password to octoprint as well. And there is no way anyone has my API key. I didnt use the default passwords.

foosel, 2016-11-06:
The saga continues in #1578 (again, please do not open multiple tickets for the same issue - that produces overhead to manage all those tickets and the information contained therein that subtracts from time available to solve issues and/or help with support requests, which is unfair towards other users who play by the rules)

closed 2016-11-06 by foosel

nophead, 2016-11-06:
Are you prone to sleep walking?